robotology-playground/WBI-Toolbox #49: WBI not compiling

robertocalandra, 2015-01-08:
When compiling the codyco-superbuild (pulled today 08/01/15 at 16.00 ) the WBIToolbox fail to find a Matlab installation and therefore terminate with error.

The version of matlab installed on the system in 2012a.

This is the relevant error:

[ 67%] Skipping update step for 'WBIToolbox'
[ 69%] Performing configure step for 'WBIToolbox'
Not searching for unused variables given on the command line.
loading initial cache file /home/icub/software/src/codyco-superbuild/build/main/WBIToolbox/CMakeFiles/YCMTmp/WBIToolbox-cache.cmake
MATLAB_ROOT = TRUE
MATLAB_MEXFILE_EXT =
CMake Error at libraries/cmake/FindMatlab.cmake:178 (STRING):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  libraries/CMakeLists.txt:11 (find_package)

CONFIGURING FOR MATLAB 32 BITS
MATLAB_MEX_LIBRARY FOUND IN = MATLAB_MEX_LIBRARY-NOTFOUND
MATLAB_MX_LIBRARY  FOUND IN = MATLAB_MX_LIBRARY-NOTFOUND
MATLAB_ENG_LIBRARY FOUND IN = MATLAB_ENG_LIBRARY-NOTFOUND
CMake Error at libraries/CMakeLists.txt:21 (message):
  Matlab not found

traversaro, 2015-01-08:
In which directory Matlab is installed? The CMake FindMatlab.cmake file searches on a handful of directories in which Matlab is usually installed (see https://github.com/robotology-playground/WBI-Toolbox/blob/master/libraries/cmake/FindMatlab.cmake#L114 ). If you installed Matlab in an unusual directory, you can point CMake to that directory by setting the MATLAB_DIR enviromental variable to the directory where Matlab is installed.

robertocalandra, 2015-01-08:
Thank you Silvio, that solved the problem. 
I am closing the issue.

closed 2015-01-08 by robertocalandra

jeljaik, 2015-01-08:
I will highlight the part of the instructions in which this is described.
Probably it's not that visible.