robotology-playground/WBI-Toolbox #50: ERROR [mdlOutput] Joint limits could not be computed

robertocalandra, 2015-01-14:
When trying to use the WBI on iCubDarmstadt01 I am get the following error:

Error reported by S-function 'robotState' in 'iCub_000/Joint Angles/S-Function':
ERROR [mdlOutput] Joint limits could not be computed

I read the Issue 30, which seems related, but by following the thread I could still not find a solution that work for me. https://github.com/robotology-playground/WBI-Toolbox/issues/30

I would really appreciate any help to try to solve this issue.
-R

jeljaik, 2015-01-14:
- [ ] Modify documentation specifying `DICUBWBI_USE_EXTERNAL_TORQUE_CONTROL:BOOL=YES` when using the real robot.

jeljaik, 2015-01-14:
Solution:
- After checking that `YARP_DATA_DIRS` was set correctly, we just changed `robotName='iCubDarmstadt01'` instead of `icub` and that helped finding the right configuration file as shown next:

```bash
||| checking [/home/icub/software/share/iCub/contexts/wbit] (context YARP_DATA_DIRS)
||| checking [/home/icub/software/share/ICUBcontrib/contexts/wbit] (context YARP_DATA_DIRS)
||| checking [/home/icub/software/src/codyco-superbuild/build/install/share/codyco/contexts/wbit] (context YARP_DATA_DIRS)
||| found /home/icub/software/src/codyco-superbuild/build/install/share/codyco/contexts/wbit
||| checking [/home/icub/software/src/codyco-superbuild/build/install/share/codyco/contexts/wbit/iCubDarmstadt01.ini] (context)
||| found /home/icub/software/src/codyco-superbuild/build/install/share/codyco/contexts/wbit/iCubDarmstadt01.ini
[robotStatus::robotConfig] icub_fixed is true!
[robotStatus::robotConfig] After reading from config file, params are 
[robotStatus::robotConfig] robot name:    icub 
[robotStatus::robotConfig] urdf file:     /home/icub/software/src/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/conf/wbit/urdf/iCubGenova01/icub.urdf 
[robotStatus::robotConfig] local name:    simulink 
[robotStatus::robotConfig] head version:  2 
[robotStatus::robotConfig] legs version:  2 
[robotStatus::robotConfig] feet version:  1 
[robotStatus::robotConfig] world reference frame: l_sole 
WARNING [robotStatus::robotConfig] Configuring WBI for the real robot. Please check the jointTorqueControl module is up and running.
ERROR [robotStatus::robotConfig] The jointTorqueControl module is not running. Please launch it before running your controller.
```

- After finding the correct configuration file for your robot the new ERROR regarding the  jointTorqueControl not being found was addressed in commit https://github.com/robotology-playground/WBI-Toolbox/commit/74c472a4e9d6c37782a97dd142f67ec0bc94eadc

closed 2015-01-14 by jeljaik