robotology-playground/WBI-Toolbox #51: Self-collision in position control

robertocalandra, 2015-01-15:
When using position control ('set Angles'), there is no self-collision check (i.e., if improper angles are set, an arm might collide with the rest of the body ). Using the limits from 'Joint Limits' is not helpful either, since it is just the min/max limits of the joints.

If it doesn't exist yet, I think it would very valuable to include for position control a self-collision check based on the urdf of the icub.

traversaro, 2015-01-15:
It would definitely be an helpful feature.. unfortunately at the moment the WBI-Toolbox position control is sending the joint position directly to the low level firmware position control. 
Some kind of "collision avoidance" could be implemented using the [FCL library](https://github.com/flexible-collision-library/fcl). Code for using an URDF model in FCL is already contained in the MoveIt! library, so adapting it could be interesting https://github.com/ros-planning/moveit_core/tree/indigo-devel/collision_detection_fcl . If you have some workforce to commit to this we will be glad to help!

jeljaik, 2015-01-23:
Closing as the goal of the block is just to set joint angles, not a "collision free configuration" which is of a higher level of abstraction.

closed 2015-01-23 by jeljaik