robotology-playground/WBI-Toolbox #52: Joint Torques return 0

robertocalandra, 2015-01-15:
Trying to read the joint torques, even during a movement, always return 0. 
I tested by generating movements in both position and torque control, and neither works.

traversaro, 2015-01-15:
Are you running the wholeBodyDynamics module?

robertocalandra, 2015-01-15:
Separately on the yarpManager? No.
Is it required?

traversaro, 2015-01-15:
Yes, basically the wholeBodyDynamics module is required to estimated the joint torques, you should find it in the robotStartup application in your yarpManager.

For more information in the force control, please check http://wiki.icub.org/wiki/Force_Control . It is a bit outdated (we will update it soon) but it explains the concepts quite well.

robertocalandra, 2015-01-15:
Ok thank you. In that case, if i understand correctly when reading the Joint Torques block there is quite a bit of processing (e.g., estimation of external forces). Is it possible to directly access the low level readings from the joint torque sensors?

traversaro, 2015-01-19:
As far as I know, `iCubDarmstad01` is measuring the joint torques through the 6 6-axis FT sensors embedded in the iCub body. You can access this sensors directly using the /icub/deviceId/analog:o port, where deviceId is one of the following `left_arm`,`right_arm`,`left_leg`,`right_leg`, `left_foot`, `right_foot`.

jeljaik, 2015-01-23:
I will  assume that after launching `wholeBodyDynamicsTree` the block for retrieving joint torques was not returning 0 anymore. Therefore I proceed to close this issue. If this is  not the case, feel free to reopen :)

closed 2015-01-23 by jeljaik

robertocalandra, 2015-01-23:
Sorry, I lost track of this issue. Yes, using the wholeBodyDynamicsTree solved the problem.