robotology-playground/WBI-Toolbox #53: After using torque control the robot remain in torque control mode

robertocalandra, 2015-01-15:
After using the torque control, the joint are unresponsive (also to restarting the robotInterface).
It took me a while to notice from the robotMotorGui that the joints were in hardware fault...
Would it make sense to automatically switch back to position control at the end of the execution?

jeljaik, 2015-01-15:
The normal behavior of the WBI-Toolbox is exactly that of leaving the robot
in the position control mode. I think you will need to provide further
information to understand if this is caused by our code or not. Nonetheless
we'll check tomorrow. Is it replicable by just trying to set torques with
the corresponding block?

randaz81, 2015-01-15:
No, I think that the wholeBodyDynamics(Tree) module, which is responsible
for joint torque estimations, was terminated (or crashed, or disconnected
from the network for more than 100milliseconds) BEFORE closing the toolbox.
As indicated in the control mode specifications, all these conditions
trigger an hardware fault (in order to protect joints from a sudden jerky
movement), which can be cleared only from the robotMotorGui or turning off
the control boards: as you reported, restarting robotInterface is not
sufficient.

traversaro, 2015-01-15:
For more information on the iCub control modes, please check: http://wiki.icub.org/wiki/Control_Modes

serena-ivaldi, 2015-01-18:
just to be sure, whenever we use WBI we have to use it with wholeBodyDynamicsTree?

DanielePucci, 2015-01-18:
The WBIToolbox is, at all intents and purposes, an interface. So, you can use it as you would use the WBI. Then,  wholeBodyDynamicsTree must be up only if  modules requiring the information from it are running.

traversaro, 2015-01-19:
In a nutshell: if you need to measure/control torques (readTorques,setTorques) the wholeBodyDynamics/wholeBodyDynamicsTree should be up.

closed 2015-02-19 by robertocalandra