robotology-playground/WBI-Toolbox #56: gains for impedance control

azadm, 2015-01-19:
The proposed gains for impedance control don't work properly in simulation. By impedance control I mean the file in ````wholeBodyImpedance```` folder. Does anyone know of better gains?
The current proposed gains are: ```` diag([[10 6 6] 2*ones(1,10) 0.5*ones(1,12)]) ````

DanielePucci, 2015-01-22:
Better with respect to what? Have you tried to do some tuning on your own?

azadm, 2015-01-23:
Yes I tried to tune them but it was unsuccessful. These gains actually did not work on my machine and made the robot unstable.
By the way, I haven't been able to run this controller with the new version of wbitoolbox yet. I will try to run this again once my problem with some of the blocks (`holder` and `baseVelocity`) is solved.

DanielePucci, 2015-02-05:
Try again and open this issue in 
https://github.com/robotology-playground/WBI-Toolbox-controllers  
if problems still arise.

closed 2015-02-05 by DanielePucci