robotology-playground/WBI-Toolbox #57: Error running torque balancing controllers

azadm, 2015-01-19:
This is the error that I get when I try to run ````torqueBalancing```` and ````torqueBalancingV1````:

````
Error evaluating 'InitFcn' callback of block_diagram 'controllerSharedMem'.

Caused by:
Undefined function or variable 'initSharedMem'.
````

Does anyone know where this function is called and where I can find this function?

traversaro, 2015-01-19:
@azadm are you still using the master branch (v0.1)? If you wait this night we will merge in master the new version of the WBI-Toolbox (see https://github.com/robotology-playground/WBI-Toolbox/issues/55 ). The new version features several improvements, and people will be able to help you, given that here at IIT everyone in now using the newer version.

azadm, 2015-01-20:
I updated to the new version of WBI-toolbox. It seems to me that torque balancing controllers are not compatible with the new version as I get the same error for all of them:
````
In an assignment A(I) = B, the number of elements in B and I must be the same.
````
and
````
Operands to the || and && operators must be convertible to logical scalar values.
````

DanielePucci, 2015-01-20:
May you provide further details?

azadm, 2015-01-20:
might be a problem with the installation of wbi toolbox because I get this error
````
Error evaluating parameter 'Value' in 'test1/Joint Angles/Input Param. for forwardKinematics and computeJacobians 1'

Caused by:
Undefined function or variable 'ROBOT_DOF'.
````
when I just want to read joint angles using joint angles block.

DanielePucci, 2015-01-20:
We kindly ask you to read the documentation on how to use the WBIToolbox at 

https://github.com/robotology-playground/WBI-Toolbox/blob/dd77704262242963fd67835e63ac04c2425f1f23/README.md

In the case unclear things arise, do not hesitate to complete it or ask us further clarifications.

azadm, 2015-01-21:
Thanks @DanielePucci 
The earlier problem that I mentioned is due to the `baseVelocity` block.

azadm, 2015-02-05:
Now I am using the new version of WBI and the error is still there.
When I try to run `torqueBalancing` and `torqueBalancingV1` I get this error:
````
Error evaluating 'InitFcn' callback of block_diagram 'controllerSharedMem'.

Caused by:
Undefined function or variable 'initSharedMem'.
````

The model is looking for `initSharedMem` as asked in the model callbacks but it can not find it.

DanielePucci, 2015-02-05:
I don't know if you noticed that we have moved all controllers in a new repository:

https://github.com/robotology-playground/WBI-Toolbox-controllers

Here you'll find new versions of the controllers. For instance, only two balancing controllers are present. Notice that these controllers have ROBOT_DOF = 23 because we seldom use the pronosupination DOFs in torque mode. So, to use them, you have just to set the variable wbi_id_list as follows
~~~
wbi_id_list    ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP
~~~
You can change this variable in the file, for instance, 
~~~
codyco-superbuild/build/install/share/codyco/contexts/wholeBodyInterfaceToolbox/wholeBodyInterfaceToolbox.ini
~~~

The gains associated with the balancing controllers. however, have been tuned for the real robot iCubGenova01. So, the tuning for making them work on the simulator is left to the user.

DanielePucci, 2015-02-05:
So, I'll close this issue since it's better to post WBI-controllers related issues in its associated repository.

closed 2015-02-05 by DanielePucci