robotology-playground/WBI-Toolbox #60: Problem with yarpWholeBodyInterface

azadm, 2015-01-21:
When I run the model `$CODYCO_SUPERBUILD_DIR/main/WBIToolbox/controllers/tests/COMPoseTesting.mdl` with iCub on the Gazebo simulator I get this error in matlab:
```
Error reported by S-function 'robotState' in 'COMPoseTesting/COMFwdKin /S-Function':
impossible to read from configuration file. Maybe yarpWholeBodyInterface was not installed? or the configuration file is empty.
```
and this in the terminal that matlab is running:
```
mdlInitializeSize >> BLOCK TYPE IS: 2
||| did not find yarpWholeBodyInterface.ini
yarp: cannot read from yarpWholeBodyInterface.ini
||| did not find yarpWholeBodyInterface.ini
yarp: cannot read from
```
This file is in `CODYCO_SUPERBUILD_DIR/install/share/codyco/robots/icubGazeboSim` and is not empty.
I have `YARP_DATA_DIRS=${YARP_DATA_DIRS}:${CODYCO_SUPERBUILD_DIR}/install/share/codyco` in my `.bash_profile` and I also have defined `$YARP_ROBOT_NAME=icubGazeboSim` in that file.

DanielePucci, 2015-01-21:
Before launching the simulation, e.g. in your init.m file of the simulink model if  you have it, type in the MATLAB command shell 

setenv('YARP_ROBOT_NAME', 'icubGazeboSim');

This should work.

azadm, 2015-01-21:
Yes, that worked. Thanks @DanielePucci 
Do you know of any permanent solution for that?

DanielePucci, 2015-01-21:
We still don't know any permanent solution. We are working on it.

traversaro, 2015-01-21:
Can you paste your complete `.bashrc`?  Enclose it in three tilde (~~~), as in:
~~~
this text area
~~~
This simplifies reading it.

azadm, 2015-01-21:
```
export Qt5_DIR=/Applications/Qt/5.3/clang_64
export PKG_CONFIG_PATH=/usr/X11/lib/pkgconfig/:$PKG_CONFIG_PATH
export GDK_PIXBUF_MODULEDIR="/usr/local/lib/gdk-pixbuf-2.0/2.10.0/loaders"
export YARP_ROOT=/Users/azadm/Documents/yarp
export YARP_DIR=/Users/azadm/Documents/yarp/build
export PATH=$YARP_DIR/bin/Release:$PATH
export YARP_DATA_DIRS=/Users/azadm/Documents/yarp/build/share/yarp:/Users/azadm/Documents/icub-main/build/share/iCub
export OpenCV_DIR=/usr/local/Cellar/opencv/2.4.9
export IPOPT_DWN_DIR=/Users/azadm/Documents/Ipopt-3.11.9
export IPOPT_DIR=/Users/azadm/Documents/Ipopt-3.11.9/build
export ICUB_ROOT=/Users/azadm/Documents/icub-main
export ICUB_DIR=/Users/azadm/Documents/icub-main/build
export PATH=$ICUB_DIR/bin/Debug:$PATH
export CODYCO_SUPERBUILD_ROOT=/Users/azadm/Documents/codyco-superbuild
export CODYCO_SUPERBUILD_DIR=/Users/azadm/Documents/codyco-superbuild/build
export PATH=$CODYCO_SUPERBUILD_ROOT/build/install/bin:$PATH
export DYLD_LIBRARY_PATH=$CODYCO_SUPERBUILD_ROOT/build/install/lib:$DYLD_LIBRARY_PATH
export YARP_DATA_DIRS=$YARP_DATA_DIRS:$CODYCO_SUPERBUILD_ROOT/build/install/share/codyco
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:/Users/azadm/Documents/gazebo-yarp-plugins/build
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:/Users/azadm/Documents/icub-gazebo
export PATH=$PATH:${CODYCO_SUPERBUILD_DIR}/install/bin/
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:${CODYCO_SUPERBUILD_DIR}/install/lib
export YARP_DATA_DIRS=${YARP_DATA_DIRS}:${CODYCO_SUPERBUILD_DIR}/install/share/codyco
$YARP_ROBOT_NAME=icubGazeboSim
```

jeljaik, 2015-01-22:
The way you set the environmental variable `$YARP_ROBOT_NAME=icubGazeboSim` is wrong. You should change that last line into:
`export YARP_ROBOT_NAME="icubGazeboSim"`

In conclusion, there's nothing wrong with the WBI-T at this point. That's just how it works and how it's explained in the README.md. You **have** to set this environmental variable with the name of the robot you wanna use (e.g. icubGazeboSim, iCubGenova01, iCubHeidelberg01, etc). The method @DanielePucci mentioned before about setting it in Matlab is just another way to do so, but it's a local change of the environmental variable, meaning, outside MATLAB's scope, that variable does not exist. For robustness I suggest that you set it in your `bashrc` and then overwrite it in your particular model (using setenv() ) according to the particular implementation.

closed 2015-01-22 by jeljaik