robotology-playground/WBI-Toolbox #66: torqueBalancingV3- error after update

yoshijan, 2015-01-27:
Hello everyone,
i use the WBI - Toolbox with gazebo and icub (current versions, ROBOT_DOF =25).
After the latest update i followed the new Readme instructions and got the COMPoseTesting working.

But if i want to run the torqueBalancingV3 controller i got a error message:
Error in port widths or dimensions. Output port 1 of 'balancingController_QP_twoFeet/Subsystem/Subsystem1/WBI to yarp3/WBI ordered' is a [1x25] matrix.

see pic below
the WBI to yarp3 just uses 23 values?!
Any ideas how to solve this issue?
Best regards
Jan
[image]

francesco-romano, 2015-01-27:
The new wholeBodyInterface configuration file by default now uses 23 DOFs. We removed the wrist prosupination as we did not control it in torque mode.
As a solution to your problem either change the `ROBOT_DOF` variable to be 23, or change the wholeBodyInterface configuration file to include again the whole arm.

Maybe @DanielePucci can give more details about this issue.

jeljaik, 2015-01-27:
The main problem here was in the block `yarp to WBI` which had been wrongly set to 23 instead of `ROBOT_DOF`. This happened after recent updates in which we have decided to control 23 DOF instead of the previous 25. For this reason the variable `ROBOT_DOF` in `initBalancingController_QP_twoFeet.m` must be set to 23. Commit https://github.com/robotology-playground/WBI-Toolbox/commit/9db1ec35c58f4746ecda7d88fdac1132daec8217 fixes this issue. Nevertheless a few more issues arise from the non-scability of the Matlab code of the controller which has a 25-DOF logic. It needs to be updated. You should also know @yoshijan that this controller was never successfully tested on the simulator but on the robot. For this reason, gains and a couple of things will not work out of the box.  If you are truly interested in this controller, you're welcome to do all the necessary changes and via a pull request we can review it and integrate in our repo. Otherwise, as soon as we get the chance we'll get back to this controller in more depth.

jeljaik, 2015-01-27:
Created issue #67 to track this problem.
Closing since the WBI-Toolbox related issue has been solved.

jeljaik, 2015-01-27:
Also @yoshijan for future issues, we would like you to try our newly introduced script `create_issue.sh` https://github.com/robotology-playground/WBI-Toolbox/blob/master/create_issue.sh which you can use to fill a template issue with automatically filled info that is useful for us in the debugging process.

Thanks!

jeljaik, 2015-01-28:
Sorry @yoshijan, we realized there had been a misinterpretation of this issue. I however reverted the conflicting blocks 'yarp to WBI' and 'WBI to yarp' to how they were in the past. Pushed changes in https://github.com/robotology-playground/WBI-Toolbox/commit/9665fbdf997431b741bd757369ef674fc3382cfe and https://github.com/robotology-playground/WBI-Toolbox/commit/70678101b6bc0dc48483d962c337805fbc2ca63a

This should solve this particular issue. Let us know please of further errors. 

@DanielePucci can you test this controller (or the newest version you told me about today) on your computer in simulation? At least to make sure it runs.

DanielePucci, 2015-02-05:
See https://github.com/robotology-playground/WBI-Toolbox/issues/57#issuecomment-73134333

closed 2015-02-05 by DanielePucci