robotology-playground/WBI-Toolbox #68: Sending 0 torques results in non-zero torques applied.

robertocalandra, 2015-01-28:
We are having serious difficulties in commanding in torque the iCubDarmstadt01 from the wbi.
After switching on the wholeBodyDynamicsTree, if we try to send a zero torque using the set Torque block of the wbi, the icub apply a strong force forward.

I am not sure if it is because of errors in the wholeBodyDynamicsTree, because of the low-level firmware or what else.
However, it is critical for us to fix this problem as soon as possible. Would it be possible to have ASAP a skype meeting with someone to diagnose and solve it?

For a video see https://www.dropbox.com/s/3g9cwr3bk44v5lt/VID_20150128_181413.mp4

PS: this bug happen when using the older version of the wbi (v0.1)

traversaro, 2015-01-28:
Which kind of torques are you measuring (you can read that on the robotMotorGui) at the joint at which you command a desired torques of 0 Nm?

traversaro, 2015-01-28:
However to debug the problem you can use the `demoForceControl` module, and set zero torques to a part through it, bypassing the WBI.

robertocalandra, 2015-01-28:
The wbi "read Torques" block returns torques up to 1.4Nm

robertocalandra, 2015-01-28:
True that I could use the demoForceControl to achieve zero torques. But the original purpose was to implement a PD controller for position tracking, and if the torques provided are not the ones applied, that is a major problem.

iron76, 2015-01-28:
@robertocalandra are you setting some sort of gravity compensation? The behaviour seems compatible with zero torques at the joint and no gravity compensation.

robertocalandra, 2015-01-28:
I tried with both gravity compensation and without gravity compensation and the results are qualitatively equal. 

I would however expect that even in presence of no gravity compensation the controller would still apply zero torques and therefore the torso "slowly" fall down naturally due to the gravity. This is not the case as the torso is actively pushing down.

iron76, 2015-01-28:
What happens with the `demoForceControl` module as suggested by @traversaro?

robertocalandra, 2015-01-28:
The robot seems to behave normally for the two arms. I did not tried the torso as it is marked as experimental. Should I?

iron76, 2015-01-28:
Yes please.

iron76, 2015-01-28:
@robertocalandra The video you posted refers to a trial with or without gravity compensation?

robertocalandra, 2015-01-28:
@iron76 If we go into zero torque mode (from demoForceControl), the behavior is the same (strong push forward of the torso).
Same thing also if we manually switch from the robotMotorGui to torque control.
(Both were tested without gravity conmpensation)

I do not remember if the gravity compensation was activated or not in the video.

randaz81, 2015-01-28:
could you please tell me the measured joint torques of the 3 joints of the
torso, as shown in the robotMotorGui, when the torso is in controlled in
position 0,0,0

iron76, 2015-01-28:
@robertocalandra, in any case the zero torque control is definitively quite a a risky control mode and I would recommend doing tests in impedance.

randaz81, 2015-01-28:
PS: ...not only the three torques when in 0,0,0 but also when the joints
are in the following positions:
20, 0, 0
0, 20, 0
0, 0, 20
Long time ago, the yarp wrapper was swapping the torques of two of these
joints. I just want to be sure that you are using a recent version of yarp.

DanielePucci, 2015-01-28:
I do agree with @iron76 that the zero-torque control is quite risky, and should be avoided even for debugging. If the torso strongly pushes forward, however, there might be a problem at low level. A possible issue might be due to the coupling between the (desired) joint torques and the applied PWMs to the motors 0B3M0 and 0B3M1 (see http://wiki.icub.org/wiki/ICub_coupled_joints#Torso). Probably @randaz81 may clarify this and claim whether or not the problem may be due to this coupling handling.

randaz81, 2015-01-29:
@DanielePucci Coupling is correct. Otherwise in the video the robot should move diagonally.
@robertocalandra Waiting for the torque measurements.

robertocalandra, 2015-01-29:
@randaz81 Here there are the torque measurementes are in the robotMotorGui (using wholeBodyDynamicsTree), averaged over 5 seconds.
Reading the torques from the wbi led to the same results, but in inverted sequence.

[0,0,0] ->[0.01, -0.2 -4.4]
[20,0,0] -> [-0.19,1.34,-4.0]
[0,20,0] -> [1.7 -4.2 -4.0]
[0,0,20] => [0.1,-0.2,-8.3]

traversaro, 2015-01-29:
Could be an issue with wholeBodyDynamicsTree ? Which options are you using? You obtain similar results with the old wholeBodyDynamics?

robertocalandra, 2015-01-29:
@traversaro I cannot test it today, as the robot is being used for other experiments, I will let you know tomorrow

robertocalandra, 2015-01-30:
@traversaro These are the results using the wholeBodyDynamics:

[0,0,0] -> [0.03, -0.18, -4.3]
[20,0,0] -> [-0.13, 1.41, -4.00]
[0,20,0] -> [1.78, -4.19, -4.11]
[0,0,20] -> [0.17, -0.15, -8.33]

I would say that they are qualitatively similar.

Question: the torso skin is currently dismounted (because of a faulty connection), as shown also in the video, could that be the problem?

robertocalandra, 2015-01-30:
PS: I am using the options --headV2 --autoconnect

randaz81, 2015-01-30:
I tried the following setup:

robot name: iCubGenova04, mechanics revision V1, ETH communication
software: Yarp, icub-main HEAD revisions + latest available firmware
module: WholeBodyDynamics (from icub-main), with and without gravity
compensation.
test:
1- estimated torque values match (with an acceptable tolerance) with the
torques provided by Roberto.
2- demoForceControl is able to control the torso smoothly in both
zero-torque and impedance mode, no issues found.

randaz81, 2015-01-30:
Before proceeding any further, I recommended to sync your robot with our test case, with a general update of yarp, icub-main, icub-firmware-shared. This to eliminate the possibility that you are stuck to a bugged old revision of something.
Please note that updating icub-main will enforce an update the control board firmware. 
If you need we can guide you remotely (>= Tuedsay 03/02/2015)

robertocalandra, 2015-01-30:
If so, yes, let's proceed with updating yarp, icub-main, etc; and yes, I would definitely like to have remote support (especially for the bontrol board firmware).
Do you have any availability *before*  03/02/2015 ?

robertocalandra, 2015-02-02:
@randaz81 Please let me know the earliest availability for remote support.

robertocalandra, 2015-02-04:
I can confirm that after the update the issue is still present: Using zero torque control (with gravity compensation and wholebodydynamics modules running) causes a fast bending of the torso forward.

Additionally after updating yarp and icub-main 
1) whoelBodydynamics is not running anymore and return the following error:
`symbol lookup error: /home/icub/software/src/yarp/build/lib/libYARP_dev.so.1: undefined symbol: _ZTIN4yarp2os3idl12WirePortableE`

2) the WBI toolbox is not running anymore and return the following error:
`Invalid MEX-file '/home/icub/software/src/codyco-superbuild/build/install/mex/robotState.mexa64': /home/icub/software/src/yarp/build/lib/libYARP_math.so.1: undefined symbol: _ZN4yarp2os4TypeC1Ev`

traversaro, 2015-02-04:
Have you updated (or at least recompiled) codyco-superbuild after updating
yarp and icub-main ? Given that this is research software, we are not
preserving ABI (http://en.wikipedia.org/wiki/Application_binary_interface)
between different commits.

robertocalandra, 2015-02-04:
Yes, of course I did recompile the codyco-superbuild.

traversaro, 2015-02-04:
Mhh then it is possible that there are two different versions of YARP floating in the system... perhaps the codyco-superbuild is building an incompatible version of YARP?
Can you paste the content of codyco-superbuild `CMakeCache.txt`?

robertocalandra, 2015-02-04:
~~~
# This is the CMakeCache file.
# For build in directory: /home/icub/software/src/codyco-superbuild/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//The threading library used by boost-thread
BOOST_THREAD_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpthread.so

//Build the testing tree.
BUILD_TESTING:BOOL=ON

//Path to a program.
BZRCOMMAND:FILEPATH=BZRCOMMAND-NOTFOUND

//The directory containing a CMake configuration file for Boost.
Boost_DIR:PATH=Boost_DIR-NOTFOUND

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory
Boost_LIBRARY_DIR:PATH=/usr/lib/x86_64-linux-gnu

//Boost system library (debug)
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost system library (release)
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost thread library (debug)
Boost_THREAD_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_thread.so

//Boost thread library (release)
Boost_THREAD_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_thread.so

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None(CMAKE_CXX_FLAGS or
// CMAKE_C_FLAGS used) Debug Release RelWithDebInfo MinSizeRel.
CMAKE_BUILD_TYPE:STRING=

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//C compiler.
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc

//Flags used by the compiler during all build types.
CMAKE_C_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release minsize builds.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during release builds (/MD /Ob1 /Oi
[... 871 lines cut ...]
_PC_TINYXML_LIBS_OTHER:INTERNAL=
_PC_TINYXML_LIBS_PATHS:INTERNAL=
_PC_TINYXML_PREFIX:INTERNAL=/usr
_PC_TINYXML_STATIC_CFLAGS:INTERNAL=
_PC_TINYXML_STATIC_CFLAGS_I:INTERNAL=
_PC_TINYXML_STATIC_CFLAGS_OTHER:INTERNAL=
_PC_TINYXML_STATIC_INCLUDE_DIRS:INTERNAL=
_PC_TINYXML_STATIC_LDFLAGS:INTERNAL=-ltinyxml
_PC_TINYXML_STATIC_LDFLAGS_OTHER:INTERNAL=
_PC_TINYXML_STATIC_LIBDIR:INTERNAL=
_PC_TINYXML_STATIC_LIBRARIES:INTERNAL=tinyxml
_PC_TINYXML_STATIC_LIBRARY_DIRS:INTERNAL=
_PC_TINYXML_STATIC_LIBS:INTERNAL=
_PC_TINYXML_STATIC_LIBS_L:INTERNAL=
_PC_TINYXML_STATIC_LIBS_OTHER:INTERNAL=
_PC_TINYXML_STATIC_LIBS_PATHS:INTERNAL=
_PC_TINYXML_VERSION:INTERNAL=2.6.2
_PC_TINYXML_tinyxml_INCLUDEDIR:INTERNAL=
_PC_TINYXML_tinyxml_LIBDIR:INTERNAL=
_PC_TINYXML_tinyxml_PREFIX:INTERNAL=
_PC_TINYXML_tinyxml_VERSION:INTERNAL=
__YCM_BOOTSTRAPPED_CALLED:INTERNAL=TRUE
__YCM_GIT_SETUP_CALLED:INTERNAL=1
__pkg_config_checked__PC_TINYXML:INTERNAL=1
~~~

traversaro, 2015-02-04:
Ok, my hypothes is wrong. Just to clarify:
 * YARP binaries work. 
 * codyco-superbuild binaries (such as WBI-Toolbox) don't work. 
 * icub-main binaries are working? you mention that wholeBodyDynamics is failing, but you also mention running it for the test? 

In any case, it can be useful to debug to paste the output of ldd /path/to/wholeBodyDynamics and ldd /path/to/yarpserver , just to understand what's different between a working binary and a not workign one.

robertocalandra, 2015-02-04:
Sorry there was a typo in the previous post:
wholeBodyDynamics is working
wholeBodyDynamicsTree is not working

traversaro, 2015-02-04:
Ok, then `ldd /path/to/wholeBodyDynamics` versus `ldd /path/to/wholeBodyDynamicsTree` would be definitely helpful.

robertocalandra, 2015-02-04:
It seems that there are two wholeBodyDynamicsTree. is it normal?

~~~
icub@tomlinson:~$ ldd /home/icub/software/src/icub-main/build/bin/wholeBodyDynamics
    linux-vdso.so.1 =>  (0x00007fff853e9000)
    libiDyn.so.0 => /home/icub/software/src/icub-main/build/lib/libiDyn.so.0 (0x00007f3387e44000)
    libYARP_OS.so.1 => /home/icub/software/src/yarp/build/lib/libYARP_OS.so.1 (0x00007f3387a62000)
    libYARP_sig.so.1 => /home/icub/software/src/yarp/build/lib/libYARP_sig.so.1 (0x00007f338782b000)
    libYARP_math.so.1 => /home/icub/software/src/yarp/build/lib/libYARP_math.so.1 (0x00007f3387616000)
    libYARP_dev.so.1 => /home/icub/software/src/yarp/build/lib/libYARP_dev.so.1 (0x00007f33872ee000)
    libYARP_init.so.1 => /home/icub/software/src/yarp/build/lib/libYARP_init.so.1 (0x00007f33870eb000)
    libskinDynLib.so.0 => /home/icub/software/src/icub-main/build/lib/libskinDynLib.so.0 (0x00007f3386ebb000)
    libiKin.so.0 => /home/icub/software/src/icub-main/build/lib/libiKin.so.0 (0x00007f3386c48000)
    libctrlLib.so.0 => /home/icub/software/src/icub-main/build/lib/libctrlLib.so.0 (0x00007f33869c8000)
    libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f338669d000)
    libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3386486000)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f33860c0000)
    libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3385dba000)
    libACE-6.0.3.so => /usr/lib/libACE-6.0.3.so (0x00007f3385a46000)
    librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f338583e000)
    libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3385620000)
    libreadline.so.6 => /lib/x86_64-linux-gnu/libreadline.so.6 (0x00007f33853d9000)
    libgsl.so.0 => /usr/lib/libgsl.so.0 (0x00007f3384f79000)
    libgslcblas.so.0 => /usr/lib/libgslcblas.so.0 (0x00007f3384d2d000)
    libipopt.so.1 => /home/icub/software/src/Ipopt/build/lib/libipopt.so.1 (0x00007f33848fd000)
    libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f33846f9000)
    libcoinmumps.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinmumps.so.1 (0x00007f3384371000)
    libcoinlapack.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinlapack.so.1 (0x00007f33840c6000)
    libcoinmetis.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinmetis.so.1 (0x00007f3383e7a000)
    libcoinblas.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinblas.so.1 (0x00007f3383c52000)
    /lib64/ld-linux-x86-64.so.2 (0x00007f33880cf000)
    libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007f3383a29000)
    libgfortran.so.3 => /usr/lib/x86_64-linux-gnu/libgfortran.so.3 (0x00007f338370f000)
    libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f33834d3000)
~~~
 
~~~
icub@tomlinson:~$ ldd /home/icub/software/src/codyco-superbuild/build/install/bin/wholeBodyDynamicsTree
    linux-vdso.so.1 =>  (0x00007fffaa1fe000)
    libYARP_OS.so.1 => /home/icub/software/lib/libYARP_OS.so.1 (0x00007fe760bf7000)
    libYARP_sig.so.1 => /home/icub/software/lib/libYARP_sig.so.1 (0x00007fe7609c2000)
    libYARP_math.so.1 => /home/icub/software/lib/libYARP_math.so.1 (0x00007fe7607ae000)
    libYARP_init.so.1 => /home/icub/software/lib/libYARP_init.so.1 (0x00007fe7605ab000)
    libidyntree.so.0.0.1 => /home/icub/software/src/codyco-superbuild/build/install/lib/libidyntree.so.0.0.1 (0x00007fe760129000)
    libwhole-body-interface.so.0.1 => /home/icub/software/src/codyco-superbuild/build/install/lib/libwhole-body-interface.so.0.1 (0x00007fe75ff1d000)
    libyarpwholebodyinterface.so.0.0.1 => /home/icub/software/src/codyco-superbuild/build/install/lib/libyarpwholebodyinterface.so.0.0.1 (0x00007fe75fbfb000)
    libskinDynLib.so.0 => /home/icub/software/lib/libskinDynLib.so.0 (0x00007fe75f9cb000)
    libkdl-codyco.so.0.1 => /home/icub/software/src/codyco-superbuild/build/install/lib/libkdl-codyco.so.0.1 (0x00007fe75f72f000)
    liborocos-kdl.so.1.2 => /home/icub/software/src/codyco-superbuild/build/install/lib/liborocos-kdl.so.1.2 (0x00007fe75f4ac000)
    libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe75f180000)
    libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe75ef6a000)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe75eba4000)
    libACE-6.0.3.so => /usr/lib/libACE-6.0.3.so (0x00007fe75e830000)
    librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe75e628000)
    libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe75e40a000)
    libreadline.so.6 => /lib/x86_64-linux-gnu/libreadline.so.6 (0x00007fe75e1c3000)
    libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe75debd000)
    libgsl.so.0 => /usr/lib/libgsl.so.0 (0x00007fe75da5d000)
    libgslcblas.so.0 => /usr/lib/libgslcblas.so.0 (0x00007fe75d810000)
    libyarpcar.so.1 => /home/icub/software/lib/libyarpcar.so.1 (0x00007fe75d60e000)
    libyarpmod.so.1 => /home/icub/software/lib/libyarpmod.so.1 (0x00007fe75d40c000)
    libiDyn.so.0 => /home/icub/software/src/icub-main/build/lib/libiDyn.so.0 (0x00007fe75d182000)
    libkdl-format-io.so.0.0 => /home/icub/software/src/codyco-superbuild/build/install/lib/../lib/libkdl-format-io.so.0.0 (0x00007fe75cf4d000)
    liburdfdom_world.so.0.3 => /home/icub/software/src/codyco-superbuild/build/install/lib/../lib/liburdfdom_world.so.0.3 (0x00007fe75cd20000)
    libiKin.so.0 => /home/icub/software/src/icub-main/build/lib/libiKin.so.0 (0x00007fe75caae000)
[... cut ...]
    libcoinmumps.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinmumps.so.1 (0x00007fdd2783c000)
    libcoinlapack.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinlapack.so.1 (0x00007fdd27591000)
    libcoinmetis.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinmetis.so.1 (0x00007fdd27344000)
    libcoinblas.so.1 => /home/icub/software/src/Ipopt/build/lib/libcoinblas.so.1 (0x00007fdd2711d000)
    libgfortran.so.3 => /usr/lib/x86_64-linux-gnu/libgfortran.so.3 (0x00007fdd26e03000)
    libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007fdd26bc7000)
~~~

traversaro, 2015-02-04:
Yes, it make sense that there are two wholeBodyDynamicsTree: the superbuild automatically installs all its subproject in the build/install directory, so you will find the same binary in the build directory and in the install directory. 

However this output highlighted the problem: the binaries of the codyco-superbuild are using the YARP libraries from `/home/icub/software/lib`, that I guess are outdated with respect to the one in `/home/icub/software/src/yarp/build/lib/`. Perhaps they are remaings of an old YARP install never uninstalled, and that path is still in `LD_LIBRARY_PATH`? I guess that now the easiest solution is to manually remove the yarp libraries in `/home/icub/software/lib`.

robertocalandra, 2015-02-09:
The issue is still open after completely reinstalling the codyco-superbuild from scratch. 
After activating wholeBodyDynamics, gravity compensation and switching to zero torque mode, there is still a drift of the torso. This time the torso is not moving forward but rotating to the side (with a torque sufficently high that a single person cannot stop it). A video is available at: https://www.dropbox.com/s/us3ibc19liamz9b/VID_20150209_170522.mp4

PS: the issue is clearly not wbi-related.

iron76, 2015-02-10:
Once more, pure torque control is a nonsense without perfect modelling (which is i.m.h.o. a chimera). It's even worse considering that you are debugging. @robertocalandra can you please do impedance control instead? Just to be more clear. Control mode should be ```Torque Control``` but choose torques as proportional to the difference between current and desired position.