robotology-playground/WBI-Toolbox #69: Memory allocation error with holder block

hu-yue, 2015-01-29:
Hi, I'm having problems with the holder block.
If I try to use it the model cannot be saved, Simulink gives an ```Memory allocation error```.
If I try to start the simulation without saving it, then  I get an ```Assertion Error``` reported by ```Stateflow```, with summary  ```object[id]!=NULL```.
This is the full output in the matlab command window when this happens:
Warning: Memory allocation error 
failed assertion in utils/object.cpp at line 2853:
	object[id]!=NULL
 ```
Stack Trace:
  [0] libmwstateflow.so:0x00007f083b880d2a(0x7f092b03e930, 4, 0x7f092b03e960 "initialize_chart_compilation", 0x7f092b03f398)
  [1] libmwstateflow.so:0x00007f083b56768c(0x7f092b03e990, 0x7f0800000005, 0x7f083bba68d8, 2)
  [2] libmwstateflow.so:0x00007f083b5619c7(0x7f092b03ea20, 0x7f092b03ebb0, 26435, 0)
  [3] libmwstateflow.so:SF::sf_mexFunction(int, mxArray_tag**, int, mxArray_tag const**)~(0x7f092b03ec80, 5, 0x7f092b03eae0, 0x127f3fdad1e1) + 102 bytes
  [4] libmex.so:mexRunMexFile~(0x7f092b03ec10, 0x7f092b03f390, 0x52b03ebe0, 0x7f092b03f2d0) + 90 bytes
  [5] libmex.so:0x00007f093519d0f9(4, 5, 0x7f092b03ec30, 0x7f09357fee09)
  [6] libmex.so:0x00007f093519df1c(0x7f0897a292a0, 6, 0x7f092b03f390, 0x52b03fe70)
  [7] libmwm_dispatcher.so:Mfh_file::dispatch_fh(int, mxArray_tag**, int, mxArray_tag**)~(0x7f09ffffffff, 0x7f0900000000, 0xffff80f6d4fc0fc1, 0x7f0825a9ecd0) + 594 bytes
  [8] libmwm_interpreter.so:0x00007f093e491bf6(0x7f0825a982a8, 0x7f093cdc6340, 0x7f0825a9c010, 0x25a982a0)
  [9] libmwm_interpreter.so:0x00007f093e49237a(0x7f0825a9eb10, 0, 0x7f092b040250, 0x7f092b03f740)
  [10] libmwm_interpreter.so:0x00007f093e492eea(0x7f0825a9eb10, 0, 0x7f092b040250, 0x7f092b0403a8)
  [11] libmwm_interpreter.so:0x00007f093e2f5bbd(0x7f082515ef34, 0x7f093e8e4630, 0x7f093e81f870, 0x1524bd80e0)
  [12] libmwm_interpreter.so:0x00007f093e321412(0x7f092b0403a8, 0x7f092b0403a4, 0x7f0825a97100, 0x7f0825568480)
  [13] libmwm_interpreter.so:0x00007f093e32153f(0x10006b900000003, 0x7f092b0402f0, 0x7f0825b15b20, 0xffffffff00000001)
  [14] libmwm_interpreter.so:0x00007f093e43e500(0, 2, 0, 0x7f0824bd8a20)
  [15] libmwm_interpreter.so:0x00007f093e25a8ac(0x7f083ede1570, 0x7f092b040b30, 0x7f092b040b38, 0x7f0860a22bb0)
  [16] libmwm_interpreter.so:0x00007f093e256993(0x7f085f51b430, 0x7f085f519de0, 63, 0x7f0900000122)
  [17] libmwm_interpreter.so:0x00007f093e257797(0, 0, 0, 0)
  [18] libmwm_interpreter.so:0x00007f093e2c2e50(0x1000900000007, 0x7f094037c4a8, 0x7f092b041e70, 0x400000000)
  [19] libmwm_dispatcher.so:Mfh_file::dispatch_fh(int, mxArray_tag**, int, mxArray_tag**)~(0, 0x7f0860b349a0, 0x1bb800000004, 0x7f092b041ec0) + 594 bytes
  [20] libmwm_interpreter.so:0x00007f093e283778(0, 0, 0, 0x7f092b0415b0 "get_sf_block_port_info")
  [21] libmwm_interpreter.so:0x00007f093e2831f9(0x7f0900000001, 0x01000000, 0x7f0825c37cee "ox/stateflow/stateflow", 0x7f092b041e01)
  [22] libmwm_interpreter.so:inCallFcnWithTrap~(0x7f092b041ec0, 0x100000000, 1, 0x7f092b041ec0) + 92 bytes
  [23] libmwm_interpreter.so:0x00007f093e2480fb(0x7f0900000001, 0x7f091dbf55b0, 0, 1)
  [24] libmwm_interpreter.so:inCallFcnWithTrapInDesiredWS(int, mxArray_tag**, int, mxArray_tag**, char const*, bool, inWorkSpace_tag*)~(0x7f091dbf55b0, 0x7f092b04154c, 64, 0x127f010900000000) + 104 bytes
  [25] libmwm_interpreter.so:inMexCallMATLAB~(0x625f66735f746567, 0x726f705f6b636f6c, 0x6f666e695f74, 0x7f092b041e60) + 220 bytes
  [26] libmwstateflow.so:0x00007f083b781ed8(0, 0, 0x65746176697250, 0)
  [27] libmwstateflow.so:SF::sf_mexFunction(int, mxArray_tag**, int, mxArray_tag const**)~(0, 0x100000006, 0x7f092b041730, 0x127f3fdad1e1) + 102 bytes
  [28] libmex.so:mexRunMexFile~(0, 0x7f092b041e60, 0x600000000, 0x7f092b041ec0) + 90 bytes
  [29] libmex.so:0x00007f093519d0f9(0, 0, 0, 0)
  [30] libmex.so:0x00007f093519df1c(0x1000000000007, 0x7f094037c4a8, 0x7f092b041e60, 0x600000000)
  [31] libmwm_dispatcher.so:Mfh_file::dispatch_fh(int, mxArray_tag**, int, mxArray_tag**)~(0x7f092b041c60, 0x7f0897c5dde0, 6, 0x7f092b041ec0) + 594 bytes
  [32] libmwm_interpreter.so:0x00007f093e283778(0x7f091dbf55b0, 0x7f08332e5c66 "sf", 0x7f092b041e10, 0x7f092b041ec0)
  [33] libmwm_interpreter.so:0x00007f093e248058(0x7f091dbf55b0, 0x7f092b041d4c, 0x7f092b041cf0, 0x7f091dbf55b0)
  [34] libmwm_interpreter.so:inCallFcnInDesiredWS~(0x7f0900000001, 0x7f092b041db0, 0xffffffff00000001, 0x100000000) + 90 bytes
  [35] libmwsl_services.so:CMatlabCommand::execute(bool, mxErrorMode)~(0x7f0884d1c6f0, 1, 0x7f092b041ec0, 6) + 111 bytes
  [36] libmwsl_services.so:slsvFEVAL_NoThrow(int, mxArray_tag**, int, mxArray_tag**, char const*, void*, bool, mxErrorMode)~(0x7f0900000001, 0x7f08ffffffff, 0x100000002, 0x7f08332e5c66 "sf") + 64 bytes
  [37] libmwsl_services.so:slsvFEVAL_NoThrow(int, mxArray_tag**, int, mxArray_tag**, char const*, bool)~(0, 0, 0x7f085f51b270, 0x7f085f51aef0) + 90 bytes
  [38] sf_sfun.mexa64:0x00007f08332e31e9(0, 0, 0, 0)
  [39] sf_sfun.mexa64:mexFunction~(0, 0x100000004, 0x7f092b042090, 0x127f3fdad1e1) + 453 bytes
  [40] libmex.so:mexRunMexFile~(0, 0x7f0825c379c0, 0x400000000, 0x7f0825b1e818) + 90 bytes
  [41] libmex.so:0x00007f093519d0f9(0x7f0937dbdb60, 0x7f0937dbdb80, 0, 0x7f083fd2ad72 "/usr/local/MATLAB/R2013a/toolbox..")
  [42] libmex.so:0x00007f093519df1c(0x7f092b042760, 0x7f092b042750, 0x7f0825c379c0, 0x400000001)
  [43] libmwm_dispatcher.so:Mfh_file::dispatch_fh(int, mxArray_tag**, int, mxArray_tag**)~(0x7f0825c379c0, 0x400000000, 0x7f091c2539a0, 2) + 594 bytes
  [44] libmwm_dispatcher.so:mdDoMatlabFcnCall(int, mxArray_tag**, int, mxArray_tag const**, Mfh_MATLAB_fn*, mpLmgrUse)~(0, 0, 0, 0x7f0825b1e7d0) + 152 bytes
  [45] libmwsl_services.so:CMatlabCommandNoWatermark::execute(bool, mxErrorMode)~(0x7f087ff238f0, 1, 0x7f0825b1e818, 4) + 73 bytes
  [46] libmwsimulink.so:0x00007f087f10ec85(0xffffffff, 0x7f091c000020, 0x7f092b0426d0, 0x7f0937b93579)
  [47] libmwsimulink.so:0x00007f087f6d03e1(0, 0, 0x7f092b043210, 0x7f0825b1e7d0)
  [48] libmwsimulink.so:0x00007f087f6d2060(46, 0x7f091c000020, 551936, 24)
  [49] libmwsimulink.so:0x00007f087f6c3458(0x7f092b043350, 0x7f087f6c0c06, 0x7f09406203f8, 0x7f091dbf55b0)
  [50] libmwsimulink.so:0x00007f087f6c6d94(0, 0, 0x7f0900000002, 0x7f0825c37210)
  [51] libmwsimulink.so:0x00007f087f6c765a(0x7f092b0434e0, 0x7f0862452d70, 0x7f092b043398, 0)
  [52] libmwsimulink.so:0x00007f087f6919cf(0x7f0824f7d1f0, 0, 0, 0x7f092b0434f8)
  [53] libmwsimulink.so:0x00007f087f43a58a(0x1007f092b0436a0, 0x7f092b043470, 0x7f092b0434f0, 38)
  [54] libmwsimulink.so:0x00007f087f48b90f(0x7f092b043670, 0x7f083c233f10, 0x7f092b043670, 0x7f092b043610)
  [55] libmwsimulink.so:0x00007f087f48c39d(0x7f092b043720, 0x7f087f7ea5b3, 0x7f0825217ca0, 0xbff0000000000000)
  [56] libmwsimulink.so:0x00007f087eb29148(24, 0xbff0000000000000, 0x40acfc0030000000, 4)
  [57] libmwsimulink.so:0x00007f087eb2e780(58, 33, 0x7f08241ad210, 0x7f09406203e0)
  [58] libmwsimulink.so:0x00007f087eb307e1(0x1007f092b043b30, 0x7f092b043900, 0x7f092b043980, 38)
  [59] libmwsimulink.so:0x00007f087f48b955(0x7f087ffe9648, 48, 0x7f087ffe9600, 0x7f092b043aa0)
  [60] libmwsimulink.so:0x00007f087f48c39d(0x7f092b043bb0, 0x7f087f7ea5b3, 0x7f0825227f40, 0xbff0000000000000)
[... 21 lines cut ...]
  [82] libmwm_dispatcher.so:0x00007f093e9ca427(0, 0x7f091c000020, 48, 1)
  [83] libmwm_interpreter.so:inFullFevalFcn~(0x7f092b046800, 0x7f094037a1ca, 0x7f091c3130e0, 0x7f091c3130e0) + 1045 bytes
  [84] libmwm_dispatcher.so:0x00007f093e9cda6a(0x7f092b046800, 0x1007f0900000002, 0x00603050 " $@", 0)
  [85] libmwm_dispatcher.so:Mfh_MATLAB_fn::dispatch_fh(int, mxArray_tag**, int, mxArray_tag**)~(0x7f092b046550, 0x7f091c4d9db0, 2, 0x7f092b046810) + 488 bytes
  [86] libmwm_interpreter.so:0x00007f093e283778(0x7f0863644920, 0x7f093bed064f, 0x7f0863b7cb10, 0x7f087c621f7e "feval")
  [87] libmwm_interpreter.so:0x00007f093e2831f9(0x7f0800000000, 0x7f0825789030, 0x7f0896bd1df0, 0x7f082574ec00)
  [88] libmwm_interpreter.so:inCallFcnWithTrap~(0x7f092b046780, 0x7f087c621f7e "feval", 0x7f0900000001, 0x7f0800000002) + 92 bytes
  [89] libmwdastudio.so:SafeCallMatlab(int, mxArray_tag**, int, mxArray_tag**, char const*, DigManager::ShowErrorFlag)~(0x7f085f4ec3a0, 0x7f085f4edec0, 0, 0x7f0825754620) + 302 bytes
  [90] libmwdastudio.so:ActionSchema::invokeCallback(mxArray_tag*)~(0x7f092b046a00, 1, 0x7f085f4ee320, 0x7f085f4ec950) + 113 bytes
  [91] libmwdastudio.so:DigCallbackInvoker::invokeUDI(UDInterface*)~(1, 0x7f092f407499 "D;��", 0x7f0825762b60, 0x7f082574cd00) + 413 bytes
[... cut ...]
  [98] libmwbridge.so:ioReadLine(bool, _IO_FILE*, boost::optional<inWorkSpace_tag* const> const&, bool)~(0x7f092b046e00, 0x7f093f34d667, 0x7f092b04b6b0, 0x7f091e59c1a0) + 838 bytes
  [99] libmwbridge.so:0x00007f09353c4a84(0x7f093f6c7270, 0x7f091e4990d0, 0x7f091e4990e0, 0x7f091e4990e0)
  [100] libmwbridge.so:0x00007f09353ca49d(0, 0x770000006e, 0, 0x7f092b046f3f)
 ```
Have you ever had a problem like this?
I'm launching matlab with preload:
 ```
alias matlab_codyco="LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19:/usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0 matlab"
 ```

francesco-romano, 2015-01-29:
It looks very similar to https://github.com/robotology-playground/WBI-Toolbox/issues/63

jeljaik, 2015-01-29:
Looking into it.

francesco-romano, 2015-01-29:
If you are using the superbuild, which version (aka: branch) are you using?
Can you check that the superbuild and all the sub projects are properly on master?
We are having hard time trying to replicate this issue.

hu-yue, 2015-01-29:
I'm still on the new_wbi_ID, I'll switch to master and try again

jeljaik, 2015-01-29:
@hu-yue `master` has recently been merged with `new_wbi_ID`, so there's no need to stay in that branch any longer. All new commits are going in `master`.

traversaro, 2015-01-29:
@hu-yue detailed instructions on how to update from `new_wbi_ID` to `master` are contained in this comment: https://github.com/robotology/codyco-superbuild/issues/46#issuecomment-70580652 .

DanielePucci, 2015-01-29:
@hu-yue try to remove the build also.

hu-yue, 2015-01-29:
Thanks guys, after switching branch the error disappeared. =)
Closing the issue.

francesco-romano, 2015-01-29:
Let's wait for @azadm to see if the same steps solve the issue

jeljaik, 2015-01-29:
Also after you move to the `master` branch we would greatly appreciate if from now on you used the `create_issue.sh` script (Linux & Mac) to create new issues, as it serves as a template for new issues that also automatically adds information relevant to us for proper debugging. 

c.c. @robotology-playground/codyco-developers

DanielePucci, 2015-01-29:
We are experiencing this problem when copying and pasting this block. This problem is quite sinister.

azadm, 2015-02-03:
Thanks @francesco-romano 
I actually had to remove codyco-superbuild and install it again to solve this problem.

closed 2015-02-03 by jeljaik