robotology-playground/WBI-Toolbox #70: Weird crash

DanielePucci, 2015-01-30:
~~~

------------------------------------------------------------------------
          std::terminate() detected at Fri Jan 30 12:31:47 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding     : Disabled
  Current Visual     : None
  Default Encoding   : UTF-8
  GNU C Library      : 2.13 stable
  MATLAB Architecture: glnxa64
  MATLAB Root        : /usr/local/MATLAB/R2014b
  MATLAB Version     : 8.4.0.150421 (R2014b)
  Operating System   : Linux 3.2.0-4-amd64 #1 SMP Debian 3.2.65-1 x86_64
  Processor ID       : x86 Family 6 Model 60 Stepping 3, GenuineIntel
  Software OpenGL    : 1
  Virtual Machine    : Java 1.7.0_11-b21 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System      : No active display

Fault Count: 1

Abnormal termination:
std::terminate()

Register State (captured):
  RAX = 0000000000000000  RBX = 00007f9fe533c4e8
  RCX = 0000000000000000  RDX = 00007f9fe532b758
  RSP = 00007f9f16fef3a0  RBP = 00007f9f16fef760
  RSI = 00007f9fe50dffb3  RDI = 00007f9f16fef3b0

   R8 = 0000000000000000   R9 = 00007f9fe2d097c0
  R10 = 0000000000000000  R11 = 0000000000000000
  R12 = 00007f9fe534c520  R13 = 00007f9f16ff03a0
  R14 = 00007f9f16ff0380  R15 = 00007f9f16ff0760

  RIP = 00007f9fe503ef4e  EFL = 0000000000000000

   CS = 4144   FS = e561   GS = 7f9f

Stack Trace (captured):
[  0] 0x00007f9fe503ef4e    /usr/local/MATLAB/R2014b/bin/glnxa64/libmwfl.so+00966478 _ZN2fl4diag5linux12context_base12capture_dataEv+00000030
[  1] 0x00007f9fe503c424    /usr/local/MATLAB/R2014b/bin/glnxa64/libmwfl.so+00955428
[  2] 0x00007f9fe40570f5   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00868597
[  3] 0x00007f9fe40592ca   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00877258
[  4] 0x00007f9fe40593c1   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00877505
[  5] 0x00007f9fe4059743   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00878403
[  6] 0x00007f9fe40597b3   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00878515
[  7] 0x00007f9fe3683386 /usr/local/MATLAB/R2014b/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6+00385926
[  8] 0x00007f9fe36833b3 /usr/local/MATLAB/R2014b/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6+00385971
[  9] 0x00007f9fe36835de /usr/local/MATLAB/R2014b/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6+00386526
[ 10] 0x00007f9fe3683acd /usr/local/MATLAB/R2014b/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6+00387789 _Znwm+00000125
[ 11] 0x00007f9fa8a57586 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmathlinalg.so+00808326 _ZNSt6vectorIdSaIdEE14_M_fill_insertEN9__gnu_cxx17__normal_iteratorIPdS1_EEmRKd+00000294
[ 12] 0x00007f9f5bcdc3e5 /home/shared/build-x86_64/yarp/lib/libYARP_dev.so.1+00951269 _ZN9jointData14read_pidOutputERN4yarp2os3idl10WireReaderE+00000229
[ 13] 0x00007f9f5bcdd885 /home/shared/build-x86_64/yarp/lib/libYARP_dev.so.1+00956549 _ZN9jointData4readERN4yarp2os3idl10WireReaderE+00000149
[ 14] 0x00007f9f5bcd9dad /home/shared/build-x86_64/yarp/lib/libYARP_dev.so.1+00941485 _ZN9jointData4readERN4yarp2os16ConnectionReaderE+00000045
[ 15] 0x00007f9f5c0393c1 /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01188801 _ZN4yarp2os4impl20PortReaderBufferBase4readERNS0_16ConnectionReaderE+00000561
[ 16] 0x00007f9f5c037a1d /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01182237 _ZN15PortCoreAdapter4readERN4yarp2os16ConnectionReaderE+00000061
[ 17] 0x00007f9f5c01a5e6 /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01062374 _ZN4yarp2os4impl8PortCore9readBlockERNS0_16ConnectionReaderEPvPNS0_12OutputStreamE+00000102
[ 18] 0x00007f9f5c02b346 /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01131334 _ZN4yarp2os4impl17PortCoreInputUnit3runEv+00008070
[ 19] 0x00007f9f5c084ea2 /home/shared/build-x86_64/yarp/lib/libYARP_OS.so.1+01498786 _Z18theExecutiveBranchPv+00000162
[ 20] 0x00007f9f58a3f636                           /usr/lib/libACE-6.0.3.so+01037878 _ZN21ACE_OS_Thread_Adapter6invokeEv+00000166
[ 21] 0x00007f9fe2f78b50              /lib/x86_64-linux-gnu/libpthread.so.0+00027472
[ 22] 0x00007f9fe2cc27bd                    /lib/x86_64-linux-gnu/libc.so.6+00903101 clone+00000109

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.
~~~

traversaro, 2015-01-30:
Do you have any idea on how to reproduce it?

DanielePucci, 2015-01-30:
Nop :(

jeljaik, 2015-02-03:
Unless you have a way to reproduce it, I hardly see how  to debug this problem. It also may be related to some `yarp read` block. What model were you using? and please, try to use the `create_issue.sh` script next time :)

DanielePucci, 2015-02-03:
Sorry Jo. You are completely right. I was using the new version of the balancing controller that has not been pushed yet :) I think that for the time being, we may close this issue, and re-open it in the right way if it arises again. What do you think?

jeljaik, 2015-02-03:
:kissing: fine for me.

closed 2015-02-03 by jeljaik