robotology-playground/WBI-Toolbox #72: how to change icub_fixed variable?

azadm, 2015-02-03:
How do we change the icub_fixed variable in the new version?

jeljaik, 2015-02-03:
Hi @azadm 
In the new version, a configuration file for the WBI-Toolbox has been created as it can be seen in:
 https://github.com/robotology-playground/WBI-Toolbox/blob/master/libraries/wbInterface/conf/wholeBodyInterfaceToolbox/wholeBodyInterfaceToolbox.ini

This is the default configuration file of the Toolbox. In the previous version of the Toolbox (v0.1) these options were mixed in the robot-specific configuration files installed with `yarpWholeBodyInterface`. Since they're unrelated, we decided to create an independent configuration file for the Toolbox. This file is then installed as:

`$CODYCO_SUPERBUILD_DIR/install/share/codyco/contexts/wholeBodyInterfaceToolbox/wholeBodyInterfaceToolbox.ini`

which by default contains:

```bash 
robot          icubGazeboSim
localName      simulink
worldRefFrame  l_sole
robot_fixed    true
wbi_id_list    ROBOT_TORQUE_CONTROL_JOINTS
wbi_config_file yarpWholeBodyInterface.ini
```

**You should change the default options in the installed file** and not the source one.

DanielePucci, 2015-02-03:
Hi Morteza, 
in the file wholeBodyInterfaceToolbox.ini you can find the option

~~~
robot_fixed     true
~~~

This file is in the following directory 
codyco-superbuild/build/install/share/codyco/contexts/wholeBodyInterfaceToolbox/wholeBodyInterfaceToolbox.ini

Beware that if you change the installed file, any time you do 

~~~
make
~~~

in the build directory of the superbuild, the file is going to be overwritten.

azadm, 2015-02-03:
Thanks @jeljaik  and @DanielePucci 
The thing is that I want to have icub (feet fixed) with 25 DoF instead of the default icub (fixed) with 10 DoF.
Is there anything else that I should change? Because after changing `robot_fixed` to `false` in the file that you mentioned, I still get an error in Matlab if I define `ROBOT_DOF = 25;` and it works with `ROBOT_DOF = 10;` no matter which model I put in Gazebo.

jeljaik, 2015-02-03:
@azadm Can you please run the  script `create_issue.sh` in the root directory of the WBI-Toollbox from terminal and fill the issue with the description of the problem? I'm gonna close this for now since the question has been answered. For the other problem you have please fill the template issue.

closed 2015-02-03 by jeljaik