robotology-playground/WBI-Toolbox #75: After clean build all skin ports send constants i.e., 240

rueckert, 2015-02-10:
Hi, 

I hope someone can help us. After a clean reinstallation of codyco-superbuild (discussed in Issue #68) a few days ago, the skin ports all send constants, e.g., via yarp read /localPortToRead /icub/skin/left_arm only constants of 240.0 are transmitted.

Another problem due to the reinstallation is that the executeable iKinCartesianSolver is missing.  According to the instructions [1], one has to compile the repository with the ccmake flag ENABLE_icubmod_cartesiancontrollerclient=on. I can not find this flag. 

We are thankful for any advice/suggestions. 

Best, Elmar 

[1] http://eris.liralab.it/brain/icub_cartesian_interface.html

jeljaik, 2015-02-10:
Hi Elmar, 

Can you please create this issue with the `./create_issue.sh`  script? which you find in the root directory of the `WBI-Toolbox`? That will give us additional information about your installation. However, please bear in mind that the kind of problem you're talking about it unrelated to the `WBI-Toolbox`. But if you file this issue as I just mentioned before we might tell you exactly where to you can proper help (or we might even tell you what's wrong).

jeljaik, 2015-02-10:
Recreated in with further information: https://github.com/robotology-playground/WBI-Toolbox/issues/75

closed 2015-02-10 by jeljaik