robotology-playground/WBI-Toolbox #77: Error obtaining robot end effector wrenches

rbonghi, 2015-02-13:
**Description:**
I would like to use the simulink blocks: `Left leg EE Wrench` and `Right leg EE Wrench`, but I have this error when I start simulation. I does not received error in build:
```
Error reported by S-function 'robotState' in 'FrictionIdentificationImproved/Model/Left leg EE Wrench/S-Function':
Error obtaining robot end effector wrenches
```

I have configured the `wholeBodyInterfaceToolbox.ini` with:
```
robot          iCubGenova01
localName      simulink_joint_friction
worldRefFrame  l_sole
robot_fixed    false
wbi_id_list    ROBOT_TORQUE_CONTROL_JOINTS
wbi_config_file yarpWholeBodyInterface.ini
```

_____

**Further Information**:
- *System Information:* `Darwin MacBook-Pro-di-Raffaello.local 14.1.0 Darwin Kernel Version 14.1.0: Mon Dec 22 23:10:38 PST 2014; root:xnu-2782.10.72~2/RELEASE_X86_64 x86_64`
- *Working branch:* `master`
- *Current commit:* `8c08d70cee361883ed663b226d3f6c546e0324f2`
- *YARP ROOT:* ``
- *ICUB_ROOT:* ``
- *YARP_DATA_DIRS:* `:/Users/Raffaello/iit/codyco-superbuild/build/install/share/codyco`
- *YARP_ROBOT_NAME:* ``
- *Yarp Installation:* `/Users/Raffaello/iit/install/bin/yarpserver`
- *LD_LIBRARY_PATH:*`:/install/lib/`
- *LD_PRELOAD:*``
- *DYLD_INSERT_LIBRARIES:*``

jeljaik, 2015-02-13:
As I already explained in https://github.com/robotology-playground/WBI-Toolbox/issues/11 and personally to you :) this in an  upcoming feature.

rbonghi, 2015-02-13:
Yes, but I understood that the problem was only for left and right arm. I'm sorry! :)

closed 2015-02-13 by rbonghi

jeljaik, 2015-02-13:
Oh no, the problem is with the estimation of end effector wrench. It's a parametric function on the desired link.

jeljaik, 2015-02-13:
I don't know exactly what you want to do, but for the feet you can use the yarp read block to read the sensors at the ankles and do the proper transformation to project it to the end-effector. I believe you have everything you need in the Toolbox

azadm, 2016-01-05:
Hi @jeljaik,
Can you explain (or is there any document that explains) how to use yarp read to read sensors at the ankles?