robotology-playground/WBI-Toolbox #78: Error starting v0.2

robertocalandra, 2015-02-19:
Trying to run code from the WBI v0.1 on v0.2 (on a fresh installation) I get the error:
`[ERR] yarpWholeBodyActuators::openDrivers error: unable to open controlboard torsoof robot iCubDarmstadt01`
I defined `ROBOT_DOF = 25`, manually added CODYCO_SUPERBUILD_DIR, and made sure that `wholeBodyInterfaceToolbox.ini` exists for iCubDarmstadt01
Did I forgot something? Could you help me to solve it?

Below the full log
~~~
[INFO][mdlStart] STARTED
[INFO][mdlStart] Publicly stating that a new child has been born: 1
[INFO][mdlStart] The string being passed for robotName is - iCubDarmstadt01
[INFO][mdlStart] Block type mask parameter: 0.000000
[INFO][mdlStart] The link passed for parametric blocks is: na
[INFO][mdlOutputs] This block will retrieve joints angles
[INFO][mdlStart] YARP is up and running!!
[INFO][robotStatus::robotStatus] robotStatus class instantiated 1 times
[INFO][mdlStart] An object robot of type wholeBodyInterface has been created
[INFO][mdlStart] About to configure robot
[INFO][robotStatus::robotConfig] Configuring...
[INFO][robotStatus::robotConfig] wbiConfFileFullPath is: /home/icub/software/codyco-superbuild/build/install/share/codyco/robots/iCubDarmstadt01/yarpWholeBodyInterface.ini
[INFO][robotStatus::robotConfig] List of properties for yarpWholeBodyInterface was interpreted correctly
[INFO][robotStatus::robotConfig] robot_fixed is true!
[DEBUG][robotStatus::robotConfig] After reading from config file, params are
[DEBUG][robotStatus::robotConfig] robot name:            iCubDarmstadt01
[DEBUG][robotStatus::robotConfig] local name:            simulink
[DEBUG][robotStatus::robotConfig] world reference frame: l_sole
[DEBUG][robotStatus::robotConfig] WBI ID List:           ROBOT_TORQUE_CONTROL_JOINTS
[ERROR]Problems instantiating the device driver torso
[ERR] yarpWholeBodyActuators::openDrivers error: unable to open controlboard torsoof robot iCubDarmstadt01
[ERROR][robotStatus::robotConfig] Initializing Whole Body Interface!
[ERROR][mdlStart] in robotConfig
[INFO][mdlTerminate] Inside robot object 0x7f39ba2a2a20
[ERROR][robotStatus::setControlMode] Reference speeds could not be set Succesfully
[INFO][mdlTerminate] ctrl mode set
[INFO][robotStatus::~robotStatus] wbInterface in destructor: 0x7f39ba4d7320
[INFO][robotStatus::~robotStatus] wbInterface has been closed and deleted correctly. -1 to go
[INFO][mdlTerminate] robot deleted
[INFO][mdlTerminate] yarpWbiOptions deleted
[INFO][mdlTerminate] all variables set to NULL
[INFO][mdlTerminate] Left mdlTerminate
[INFO][mdlTerminate] Left mdlTerminate
[INFO][mdlTerminate] Left mdlTerminate
[INFO][mdlTerminate] Left mdlTerminate
[INFO][mdlTerminate] Left mdlTerminate
~~~

francesco-romano, 2015-02-19:
Hi @robertocalandra 
I was hoping to find some C++ examples to do some low level test, but unfortunately they move the robot so it is not "safe" to launch it.

When the error occur can you try to do a `yarp name list` in a terminal and post here the result? Does the robotMotorGui correctly work?
I assume you are working with the real robot, aren't you?

robertocalandra, 2015-02-19:
Hi @francesco-romano,

Yes, I am working with the real robot.
and Yes, the robotMotorGui work correctly.
I will post the results of `yarp name list` as soon as the robot is free (it is currently busy with other experiments)

Thank you

traversaro, 2015-02-19:
Which configuration file for the yarpWholeBodyInterface are you using?  I guess there is a `robot` parameter that should be setted to `icub` while it is setted to `iCubHeidelberg01`.

traversaro, 2015-02-19:
Long explanation: unfortunatly there is a name clash between the `YARP_ROBOT_NAME` concept (a unique name for each robot, such as `iCubHeidelberg01` , see http://wiki.icub.org/yarpdoc/yarp_data_dirs.html ) and the `robot` option, that as a convention for iCub software usually mean the string that prefixs all port created by the robot (as in `icub` in `/icub/right_leg/state:o`, see http://wiki.icub.org/wiki/Summary_of_iCub_Software_Development_Guidelines#An_Example_of_how_to_Configure_the_Module ),

Anyway properly documenting yarpWbi parameters will probably solve this problem, see https://github.com/robotology-playground/yarp-wholebodyinterface/issues/22 .

robertocalandra, 2015-02-19:
I think that I am using the `yarpWholeBodyInterface` of iCubDarmstadt01. How can I check it?
But in practice then should I change the name of the `robot` used in the simulation? or change it somewhere else?

traversaro, 2015-02-19:
The file loaded by `yarpWholeBodyInterface` is this one, from the output of WBI-Toolbox:
~~~
[INFO][robotStatus::robotConfig] wbiConfFileFullPath is: /home/icub/software/codyco-superbuild/build/install/share/codyco/robots/iCubDarmstadt01/yarpWholeBodyInterface.ini
~~~

jeljaik, 2015-02-19:
Hello @robertocalandra, your configuration file has the wrong robot name set to `iCubDarmstadt01`. Since you  are using the real platform, this should be `icub` as explained in:
https://github.com/robotology-playground/WBI-Toolbox#notes-on-configuration-files

Can you share with us the content of the file @traversaro just mentioned?

traversaro, 2015-02-19:
Can you post it? 

As far as I get it, for now we consider the real robots and the simulated robots as two different robots, with different `YARP_ROBOT_NAME`s (`iCubDarmstad01` and `icubGazeboSim`) and two different `robot` parameters (`icub` and `icubGazeboSim`).

Said that, a desirable long term goal is having a proper simulation model for each real robot. For more info on this, see: https://github.com/robotology/codyco-superbuild/issues/55 .

robertocalandra, 2015-02-21:
I changed `robot` in `wholeBodyInterfaceToolbox` to icub, but i still ge the same error.
This is the content of the two files you asked:

wholeBodyInterfaceToolbox.ini
~~~
robot          icub
localName      simulink
worldRefFrame  l_sole
robot_fixed    true
wbi_id_list    ROBOT_TORQUE_CONTROL_JOINTS
wbi_config_file yarpWholeBodyInterface.ini

~~~

yarpWholeBodyInterface.ini
~~~
robot icub
urdf model.urdf
getLimitsFromControlBoard

#Original information extracted from http://eris.liralab.it/wiki/ICub_joints
#Every "actuated" joint in the wbi is mapped to an axes of a yarp controlboard
#torso
#left_leg
#right_leg

#joints
#IMU
#FTS

[WBI_STATE_OPTIONS]
WORLD_REFERENCE_FRAME l_sole

[WBI_YARP_JOINTS]
# WBI_JOINT_ID = (YARP_CONTROLBOARD_NAME,YARP_CONTROLBOARD_AXIS)
# The port prefix of the controlboard is obtained as /${robot}/${YARP_CONTROLBOARD_NAME}
torso_yaw        = (torso,0)
torso_roll       = (torso,1)
torso_pitch      = (torso,2)

neck_pitch       = (head,0)
neck_roll        = (head,1)
neck_yaw         = (head,2)

l_shoulder_pitch = (left_arm,0)
l_shoulder_roll  = (left_arm,1)
l_shoulder_yaw   = (left_arm,2)
l_elbow          = (left_arm,3)
l_wrist_prosup   = (left_arm,4)
l_wrist_pitch    = (left_arm,5)
l_wrist_yaw      = (left_arm,6)

r_shoulder_pitch = (right_arm,0)
r_shoulder_roll  = (right_arm,1)
r_shoulder_yaw   = (right_arm,2)
r_elbow          = (right_arm,3)
r_wrist_prosup   = (right_arm,4)
r_wrist_pitch    = (right_arm,5)
r_wrist_yaw      = (right_arm,6)

l_hip_pitch      = (left_leg,0)
l_hip_roll       = (left_leg,1)
l_hip_yaw        = (left_leg,2)
l_knee           = (left_leg,3)
l_ankle_pitch    = (left_leg,4)
l_ankle_roll     = (left_leg,5)

r_hip_pitch      = (right_leg,0)
r_hip_roll       = (right_leg,1)
r_hip_yaw        = (right_leg,2)
r_knee           = (right_leg,3)
r_ankle_pitch    = (right_leg,4)
r_ankle_roll     = (right_leg,5)

[WBI_YARP_IMU_PORTS]
imu_frame /inertial

[WBI_YARP_FT_PORTS]
l_arm_ft_sensor /left_arm/analog:o
r_arm_ft_sensor /right_arm/analog:o
l_leg_ft_sensor /left_leg/analog:o
l_foot_ft_sensor /left_foot/analog:o
r_leg_ft_sensor /right_leg/analog:o
r_foot_ft_sensor /right_foot/analog:o

[WBI_ID_LISTS]
# List of joints of the joints of the robot for various uses
ROBOT_TORSO_JOINTS          = (torso_pitch,torso_roll,torso_yaw)

ROBOT_HEAD_JOINTS           = (neck_pitch, neck_roll, neck_yaw)

ROBOT_LEFT_ARM_JOINTS       = (l_shoulder_pitch, l_shoulder_roll, l_shoulder_yaw, l_elbow, l_wrist_prosup)
ROBOT_RIGHT_ARM_JOINTS      = (r_shoulder_pitch, r_shoulder_roll, r_shoulder_yaw, r_elbow, r_wrist_prosup)

ROBOT_LEFT_ARM_DYNAMIC_MODEL_JOINTS = (ROBOT_LEFT_ARM_JOINTS,l_wrist_pitch,l_wrist_yaw)
ROBOT_RIGHT_ARM_DYNAMIC_MODEL_JOINTS = (ROBOT_RIGHT_ARM_JOINTS,r_wrist_pitch,r_wrist_yaw)

ROBOT_LEFT_LEG_JOINTS       = (l_hip_pitch,l_hip_roll,l_hip_yaw,l_knee,l_ankle_pitch,l_ankle_roll)
ROBOT_RIGHT_LEG_JOINTS      = (r_hip_pitch,r_hip_roll,r_hip_yaw,r_knee,r_ankle_pitch,r_ankle_roll)

ROBOT_DYNAMIC_MODEL_JOINTS  = (ROBOT_TORSO_JOINTS, ROBOT_HEAD_JOINTS, ROBOT_LEFT_ARM_DYNAMIC_MODEL_JOINTS, ROBOT_RIGHT_ARM_DYNAMIC_MODEL_JOINTS, ROBOT_LEFT_LEG_JOINTS, ROBOT_RIGHT_LEG_JOINTS)

ROBOT_MAIN_JOINTS           = (ROBOT_TORSO_JOINTS,ROBOT_LEFT_ARM_JOINTS, ROBOT_RIGHT_ARM_JOINTS,ROBOT_LEFT_LEG_JOINTS,ROBOT_RIGHT_LEG_JOINTS )

ROBOT_TORQUE_CONTROL_JOINTS = (ROBOT_TORSO_JOINTS,ROBOT_LEFT_ARM_JOINTS, ROBOT_RIGHT_ARM_JOINTS,ROBOT_LEFT_LEG_JOINTS,ROBOT_RIGHT_LEG_JOINTS )
ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP = (ROBOT_TORSO_JOINTS,l_shoulder_pitch, l_shoulder_roll, l_shoulder_yaw, l_elbow, r_shoulder_pitch, r_shoulder_roll, r_shoulder_yaw, r_elbow,ROBOT_LEFT_LEG_JOINTS,ROBOT_RIGHT_LEG_JOINTS )

# List of imus of the robot
ROBOT_MAIN_IMUS             = (imu_frame)

# List of 6-axes ft sensors of the robot
ROBOT_LEFT_ARM_FTS          = (l_arm_ft_sensor)
ROBOT_RIGHT_ARM_FTS         = (r_arm_ft_sensor)
ROBOT_LEFT_LEG_FTS          = (l_leg_ft_sensor,l_foot_ft_sensor)
ROBOT_RIGHT_LEG_FTS         = (r_leg_ft_sensor,r_foot_ft_sensor)
ROBOT_MAIN_FTS              = (ROBOT_LEFT_ARM_FTS,ROBOT_RIGHT_ARM_FTS,ROBOT_LEFT_LEG_FTS,ROBOT_RIGHT_LEG_FTS)

~~~

robertocalandra, 2015-02-21:
Ok, I find the problem.
By changing within matlab `robotName` to `icub` made the trick.
However, in the documentation it is written that within matlab `robotName` should be the name of the robot (e.g., iCubDarmstadt01). 
Perhaps it is still old documentation?

closed 2015-02-21 by robertocalandra

DanielePucci, 2015-02-21:
Yes Roberto. Thank you very much.