robotology-playground/WBI-Toolbox #88: Error when using the block: wholeBodyModel/Jacobians/Link Jacobian

iron76, 2015-04-04:
**Description:**
The installation seems to work fine in general but the block wholeBodyModel/Jacobians/Link Jacobian is giving me the following error:
```
Error in port widths or dimensions. Output port 1 of 'untitled/Link Jacobian/Link Jacobian' 
is a one dimensional vector with 186 elements.
```

_____

**Further Information**:
- *System Information:* `Darwin Francescos-MacBook-Pro.local 14.1.0 Darwin Kernel Version 14.1.0: Thu Feb 26 19:26:47 PST 2015; root:xnu-2782.10.73~1/RELEASE_X86_64 x86_64`
- *Working branch:* `master`
- *Current commit:* `312396067ce921bbfeb330652fe61b41c01263d9`
- *YARP ROOT:* ``
- *ICUB_ROOT:* ``
- *YARP_DATA_DIRS:* `/Users/iron/Code/bin/share/yarp:/share/iCub:/Users/iron/Code/codyco-superbuild/build/install/share/codyco`
- *YARP_ROBOT_NAME:* ``
- *Yarp Installation:* `/Users/iron/Code/bin/bin/yarpserver`
- *LD_LIBRARY_PATH:*``
- *LD_PRELOAD:*``
- *DYLD_INSERT_LIBRARIES:*``

DanielePucci, 2015-04-04:
This problem seems to be due to a dimension mismatch between ROBOT_DOF - a matlab variable that you should set in your workspace - with the length of the list associated to the WBI. The main point is that we still do not provide clear information when these sizes do not match (see #38 ). Anyway, further information can be found at https://github.com/robotology-playground/WBI-Toolbox

jeljaik, 2015-04-04:
This happens as @DanielePucci has pointed out when the variable `ROBOT_DOF` is defined to something different from the DOF of the corresponding list that you have defined for the current robot. To  debug do the following: 

1. Check the configuration file of the WBI-Toolbox in `$CODYCO_SUPERBUILD_DIR/install/codyco/contexts/wholeBodyInterfaceToolbox` It should read something like this by default.
```bash
robot                  icubGazeboSim
localName         simulink
worldRefFrame  l_sole
robot_fixed         false
wbi_id_list          ROBOT_TORQUE_CONTROL_JOINTS
wbi_config_file   yarpWholeBodyInterface.ini
```
If you are using the simulator, then the value for `robot` is fine. With the last changes `robot_fixed` does not matter anymore and it will be deprecated soon. `wbi_id_list` is correct in this case if you want to control 25-DOF. If you are using one of the latest controllers this should be `ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP` which correspond to a 23-DOF iCub.  These lists names can be found in `$CODYCO_SUPERBUILD_DIR/install/share/codyco/robots/icubGazeboSim`

2. Once the configuration of the toolbox is correct, make sure that you are either setting the environmental variable `YARP_ROBOT_NAME` in `./bashrc` or from within Matlab (e.g. using `setenv('YARP_ROBOT_NAME', 'icubGazeboSim') as I believe you're doing) to for example `iCubGenova01` or `icubGazeboSim`. Depending on whether you use the real robot or the simulator. 

3. In Matlab your initialization script should define the following variables (assuming `wbi_id_list = ROBOT_TORQUE_CONTROL_JOINTS_WITHOUT_PRONOSUP`
```bash
ROBOT_DOF = 23;
localName = 'simulink';
robotName = 'icubGazeboSim';
Ts = 0.010;
```

**If you are using the real robot instead remember this variable robotName corresponds to `icub` (not `iCubGenova01`). If `ROBOT_DOF` is 25, this means you should have `wbi_id_list = ROBOT_TORQUE_CONTROL_JOINTS` in the WBI-Toolbox configuration file as explained in 1**.

Once you have checked all of this you're good to run the simulation.

jeljaik, 2015-04-13:
@iron76 Can we close this? :)

DanielePucci, 2021-04-28:
Probably yes. Closing :-) @jeljaik @iron76

closed 2021-04-28 by DanielePucci