robotology-playground/WBI-Toolbox #93: Centroidal Momentum Matrix (CMM)

azadm, 2015-05-01:
I am wondering if it is possible to add a block that calculates the CMM?

traversaro, 2015-05-01:
I guess the block is already there... @DanielePucci can provide more informations.

DanielePucci, 2015-05-01:
Yes, we have it :-)

DanielePucci, 2015-05-01:
Hold on, what do you precisely mean by Centroidal Momentum Matrix?

traversaro, 2015-05-01:
The CMM is the  "Jacobian" of the centroidal momentum.

azadm, 2015-05-01:
Yes. According to Orin, by CMM I mean the matrix that maps the generalized velocities to the momentum about the centre of mass.

DanielePucci, 2015-05-01:
Ok guys, then I do not remember and I should verify it. Probably it is not implemented since we do not use it in our balancing  controllers. I think that @jeljaik can provide you with additional information for the time being.

azadm, 2015-05-07:
Does `getCentroidalMomentumJacobian` function in `DynTree.cpp` have anything to do with the CMM? @traversaro

traversaro, 2015-05-07:
Yes the `getCentroidalMomentumJacobian` return the CMM.

azadm, 2015-05-07:
Good. So it is part of iDynTree but not yarpWholeBodyModel...

traversaro, 2015-05-08:
Yes. The wholeBodyInterface/WBI-Toolbox was always meant to be a rather simplified interface exposing basic functions available on all humanoid robots. 

However we are now working on Matlab bindings using SWIG [1] to automatically expose iDynTree classes as Matlab classes.

We have some simply prototype working, for example in https://github.com/robotology-playground/idyntree/blob/master/doc/geometric_classes.md . 

Sooner or later we will support also the advanced dynamics classes and all their methods, as the CMM computation . 

[1] http://www.swig.org/