robotology-playground/WBI-Toolbox #97: [WBI-Toolbox] Possibility of setting a world-to-base roto-translation

DanielePucci, 2015-05-27:
As far as I am concerned, the world-to-base rototranslation passed to all blocks through "q" is not taken into account, and it is handled at lower levels from specific configurations. Should we modify all blocks so that this rototranslation is actually used? @jeljaik @francesco-romano @traversaro

francesco-romano, 2015-05-27:
There is a preliminary modification into a branch (I don't remember the name, but I pushed into it 2 days ago)

The problem is that without major changes (imho) it is not possible to set the transformation. 
(The modifications are: moving xBase to a static object.  But at this point  does it make sense to maintain the full 16+n input to the blocks?)

In the branch I added a new block. If the block is added to the model, the external input to the various block should be considered (I haven't tested this part. Maybe there are issues here) 
At the current state the input to the new block is ignored. If the solution work we can remove it

Inviato da iPhone

jeljaik, 2015-06-03:
Done in https://github.com/robotology-playground/WBI-Toolbox/commit/67ddcf863fbb7aef339ec54d6285187a9f13114b

closed 2015-06-03 by jeljaik