robotology-playground/WBI-Toolbox #105: [WBI-Toolbox] Crash on Mac OsX

DanielePucci, 2015-11-03:
**Description:**
Please describe your problem here...

_____

**Further Information**:
- *System Information:* `Darwin Danieles-MacBook-Pro.local 15.0.0 Darwin Kernel Version 15.0.0: Sat Sep 19 15:53:46 PDT 2015; root:xnu-3247.10.11~1/RELEASE_X86_64 x86_64`
- *Working branch:* `master`
- *Current commit:* `3f838aa9da5952171419b61e1be41ed90d22cf22`
- *YARP ROOT:* `/Users/DanielePucci/src/yarp`
- *ICUB_ROOT:* `/Users/DanielePucci/src/icub-main`
- *YARP_DATA_DIRS:* `/Users/DanielePucci/src/install/share/yarp:/Users/DanielePucci/src/install/share/iCub:/Users/DanielePucci/src/codyco-superbuild/build/install/share/codyco`
- *YARP_ROBOT_NAME:* `iCubGenova02`
- *Yarp Installation:* `/Users/DanielePucci/src/install/bin/yarpserver`
- *LD_LIBRARY_PATH:*``
- *LD_PRELOAD:*``
- *DYLD_INSERT_LIBRARIES:*``

MATLAB crash file:/Users/DanielePucci/matlab_crash_dump.24813-1:

@jeljaik @francesco-romano 
------------------------------------------------------------------------
       Segmentation violation detected at Tue Nov  3 16:14:29 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Current Visual      : Quartz
  Default Encoding    : US-ASCII
  Host Name           : Danieles-MacBook-Pro.local
  MATLAB Architecture : maci64
  MATLAB Root         : /Applications/MATLAB_R2015b.app
  MATLAB Version      : 8.6.0.267246 (R2015b)
  OpenGL              : hardware
  Operating System    : Darwin 15.0.0 Darwin Kernel Version 15.0.0: Sat Sep 19 15:53:46 PDT 2015; root:xnu-3247.10.11~1/RELEASE_X86_64 x86_64
  Processor ID        : x86 Family 6 Model 61 Stepping 4, GenuineIntel
  Virtual Machine     : Java 1.7.0_75-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : Quartz

Fault Count: 1

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 0000000000000019  RBX = 000000014755a6e6
  RCX = 0000700000a9c980  RDX = 00007fff8c6b60dc
  RSP = 00007fff937ff7de  RBP = 0000700000a9c9a0
  RSI = 0000000000000020  RDI = 00007f8c767cb438

   R8 = 00007f8c767cb438   R9 = 0000700000a9ca68
  R10 = 0000700000a9c9d0  R11 = 00007fff93802225
  R12 = 00007f8c767cb438  R13 = 00007f8c62926380
  R14 = 00007f8c767cb438  R15 = 0000700000a9cc30

  RIP = 0000700000a9cae0  RFL = 000000014745c9c5

   CS = 0000700000a9ca00   FS = 00007fff8c6b60dc   GS = 00007f8c70d04020

Stack Trace (from fault):
[  0] 0x000000010e7bca34 /Applications/MATLAB_R2015b.app/bin/maci64/libmwfl.dylib+00027188 _ZN2fl4diag15stacktrace_base7captureERKNS0_14thread_contextEm+00000052
[  1] 0x000000010e7c035a /Applications/MATLAB_R2015b.app/bin/maci64/libmwfl.dylib+00041818 _ZN2fl4test17terminate_handledEv+00000746
[  2] 0x000000010e7bfe08 /Applications/MATLAB_R2015b.app/bin/maci64/libmwfl.dylib+00040456 _ZN2fl4diag13terminate_logEPKcPK17__darwin_ucontext+00000152
[  3] 0x00000001104e7316 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00336662 _Z32mnRunPathDependentInitializationv+00002806
[  4] 0x00000001104e75a9 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00337321 _Z32mnRunPathDependentInitializationv+00003465
[  5] 0x00000001104e58ca /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00329930 mnFatalSignalHandler+00000282
[  6] 0x00007fff9c78452a           /usr/lib/system/libsystem_platform.dylib+00009514 _sigtramp+00000026
[  7] 0x000000010c72bd2f /Applications/MATLAB_R2015b.app/bin/maci64/libicuuc.54.dylib+00081199 ucnv_io_getConverterName_54+00000095
[... 88 lines cut ...]
[ 96] 0x00000001131092e5 /Applications/MATLAB_R2015b.app/bin/maci64/libmwbridge.dylib+00062181 _Z8mnParserv+00000453
[ 97] 0x00000001104d0f92 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00245650 _ZN11mcrInstance30mnParser_on_interpreter_threadEv+00000034
[ 98] 0x00000001104b62d1 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00135889 _ZN3mcr7runtime17InterpreterThread4Impl22TypedInvocationRequestIvE29doNotRethrowMxArrayExceptionsERKN5boost8functionIFvvEEE+00000033
[ 99] 0x00000001104b6d99 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00138649 _ZN5boost3_bi5list1INS0_5valueINS_8functionIFvvEEEEEEC2ES6_+00000569
[100] 0x00000001104b6726 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00136998 _ZN5boost6detail9task_baseIvE3runEv+00000182
[101] 0x00000001104b64d7 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00136407 _ZN5boost13packaged_taskIvE10get_futureEv+00000343
[102] 0x00000001104b24e3 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00120035 _ZN3mcr7runtime17InterpreterThread4Impl26invocation_request_handlerEl+00000035
[103] 0x000000011003bb4f /Applications/MATLAB_R2015b.app/bin/maci64/libmwservices.dylib+00183119 _ZN10eventqueue18UserEventQueueImpl5flushEv+00000975
[104] 0x000000011479c325 /Applications/MATLAB_R2015b.app/bin/maci64/libmwuix.dylib+00041765 _Z25isHandleUserEventCallbackPFvPvE+00000645
[105] 0x00000001101245cc /Applications/MATLAB_R2015b.app/bin/maci64/libmwservices.dylib+01136076 _Z40svWS_ProcessPendingEventsWithLimitedRateb+00016332
[106] 0x000000011011fd62 /Applications/MATLAB_R2015b.app/bin/maci64/libmwservices.dylib+01117538 _Z25svWS_ProcessPendingEventsiib+00000898
[107] 0x000000011011fa3b /Applications/MATLAB_R2015b.app/bin/maci64/libmwservices.dylib+01116731 _Z25svWS_ProcessPendingEventsiib+00000091
[108] 0x00000001104b2717 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00120599 _ZN3mcr7runtime17InterpreterThread4Impl14process_eventsERKN5boost10shared_ptrIS2_EE+00000103
[109] 0x00000001104b2a06 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00121350 _ZN3mcr7runtime17InterpreterThread4Impl3runERKN5boost10shared_ptrIS2_EEPNS2_12init_contextE+00000358
[110] 0x000000011049b116 /Applications/MATLAB_R2015b.app/bin/maci64/libmwmcr.dylib+00024854 _Z26run_init_and_handle_eventsPv+00000054
[111] 0x00007fff978f59b1            /usr/lib/system/libsystem_pthread.dylib+00014769 _pthread_body+00000131
[112] 0x00007fff978f592e            /usr/lib/system/libsystem_pthread.dylib+00014638 _pthread_body+00000000
[113] 0x00007fff978f3385            /usr/lib/system/libsystem_pthread.dylib+00004997 thread_start+00000013

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

francesco-romano, 2015-11-03:
How many times happened?

In the stack trace I cannot see any reference to our code

DanielePucci, 2015-11-03:
All the time..

francesco-romano, 2015-11-03:
Check if there are any strange strings somewhere..

francesco-romano, 2015-11-04:
The issue seems related to the presence of a `To Workspace` block.

@DanielePucci is investigating

DanielePucci, 2015-11-04:
"I had a dog and his name was, bingo!" [1]

So, it is the simulink block `To Workspace`. Simulink crashes also if a `Scope` dumps data to the workspace.

I contacted `MATLAB` support, and they are aware of this bug.

[1]: https://www.youtube.com/watch?v=Ks7cJIIqCHY&t=1m20s

jeljaik, 2015-11-04:
Bummer!

traversaro, 2015-11-04:
So you have been actually contacted by a `technical support engineer`! Amazing!

DanielePucci, 2015-11-04:
Yep!

DanielePucci, 2015-11-05:
I've news from the MathWorks support. Apparently, on MAC OsX, the block `ToWorkspace` makes Simulink crash when the `username` is longer than ten characters. They have also provided Mac users with a patch, which can be found here:

http://www.mathworks.com/support/bugreports/1295437

I've tested it, and it works.

@francesco-romano @iron76 @nunoguedelha @vtikha @jeljaik @traversaro @claudia-lat

closed 2015-11-05 by DanielePucci