robotology-playground/WBI-Toolbox #106: All blocks should take the state as input

DanielePucci, 2015-11-09:
Some of the blocks are still internally evaluating the state, such as jacobians and forward kinematics. They should take the state as input.

@jeljaik

jeljaik, 2015-11-11:
Done for the parametric forward kinematics block. Please test and give me some feedback. For the amount of time it would require to have an additional port in this block to read the world to base roto-translation, I have left it as it was for other blocks, i.e. first sixteen elements correspond to the roto-translation matrix and the rest to the robot configuration. Please test it @DanielePucci.

jeljaik, 2015-11-11:
Changed in https://github.com/robotology-playground/WBI-Toolbox/commit/bb0d8aa2c3c0214706bc5c50aa1f1779760dc98c