robotology-playground/WBI-Toolbox #110: Set base pose block: initialisation issues

DanielePucci, 2015-11-18:
Extreme attention must be paid when using the `Set base pose` block. If it is not run as first block - by defining accurately the blocks' priorities -  at the first iteration users may have problems in evaluating quantities that depend on the base root-translation. If the model represents a control action, a workaround is to use the `enable` block on the output signal, and wait a little amount of time before writing the command signals. Clearly, this must be dealt with some other low level policy @jeljaik @francesco-romano @traversaro @gabrielenava.

jeljaik, 2015-11-18:
I will then update this in the documentation of the block.

traversaro, 2015-11-18:
I don't understand. The robot position (as part of the robot state) is not taken in input to all the blocks, as in https://github.com/robotology-playground/WBI-Toolbox/issues/106 ?

DanielePucci, 2015-11-20:
Not all of them. This is being addressed only recently, and we still have blocks that automatically retrieve the robot position unfortunately. That's why back in June, I and @francesco-romano decided to implement the ~Set base block~ as a workaround.