robotology-playground/WBI-Toolbox #111: Error

misaki43, 2016-01-12:
Hi !
I am in the step: Installing the WBI-Toolbox>>Installation.
I wrote: "~/codyco-superbuild/build/main/WBIToolbox$ make" that is supposed to allow the "startup_wbitoolbox.m" file to found correctly directories. 
But during the "make", an error appeared: 
>/home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp: In function ‘void mdlOutputs(SimStruct*, int_T)’:
/home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp:1464:68: error: ‘CTRL_DEG2RAD’ was not declared in this scope
             if(!robot->setCtrlMode (wbi::CTRL_MODE_POS, ROBOT_DOF, CTRL_DEG2RAD*10.0)) {
                                                                    ^
/home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp:1464:68: note: suggested alternative:
In file included from /usr/local/include/iCub/ctrl/adaptWinPolyEstimator.h:39:0,
                 from /home/user/codyco-superbuild/build/install/include/yarpWholeBodyInterface/yarpWholeBodyActuators.h:29,
                 from /home/user/codyco-superbuild/build/install/include/yarpWholeBodyInterface/yarpWholeBodyInterface.h:23,
                 from /home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp:24:
/usr/local/include/iCub/ctrl/math.h:72:21: note:   ‘iCub::ctrl::CTRL_DEG2RAD’
 extern const double CTRL_DEG2RAD;
                     ^
/home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp: In function ‘void mdlTerminate(SimStruct*)’:
/home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp:1991:63: error: ‘CTRL_DEG2RAD’ was not declared in this scope
             robot->setCtrlMode(wbi::CTRL_MODE_POS, ROBOT_DOF, CTRL_DEG2RAD * 0.0);
                                                               ^
/home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp:1991:63: note: suggested alternative:
In file included from /usr/local/include/iCub/ctrl/adaptWinPolyEstimator.h:39:0,
                 from /home/user/codyco-superbuild/build/install/include/yarpWholeBodyInterface/yarpWholeBodyActuators.h:29,
                 from /home/user/codyco-superbuild/build/install/include/yarpWholeBodyInterface/yarpWholeBodyInterface.h:23,
                 from /home/user/codyco-superbuild/main/WBIToolbox/libraries/wbInterface/src/wbInterface.cpp:24:
/usr/local/include/iCub/ctrl/math.h:72:21: note:   ‘iCub::ctrl::CTRL_DEG2RAD’
 extern const double CTRL_DEG2RAD;
                     ^
make[2]: *** [libraries/wbInterface/CMakeFiles/robotState.dir/src/wbInterface.cpp.o] Error 1
make[1]: *** [libraries/wbInterface/CMakeFiles/robotState.dir/all] Error 2
make: *** [all] Error 2

Can you help me please?

traversaro, 2016-01-12:
Hi @misaki43 , 
  that is a problem related to a change introduced in the `icub-main` repository (see https://github.com/robotology/QA/issues/81#issuecomment-164758967 if you are interested in the background ). 
It has been fixed in a recent commit ( https://github.com/robotology-playground/WBI-Toolbox/commit/ffbe08c521d56c02a0c4194a6fa9f5a671909794 ) so it should be sufficient to just update the repository to fix the problem. 
To update all the repositories in the codyco-superbuild, just go in the codyco superbuilld build directory and type: 
~~~
git pull
~~~
and then 
~~~
make update-all
~~~
see https://github.com/robotology/codyco-superbuild#update for more info.

misaki43, 2016-01-12:
Indeed, the problem disappear. 
Sorry, I didn't find this comment and I don't have the pull-reflex (but now i will do it).
Thanks !

closed 2016-01-12 by misaki43

traversaro, 2016-01-12:
Don't worry, in general users are not expected to read everything goes on in the repo. : ) 
Updating as soon as you get a problem to see if it the problem has been fixed in the latest version is nevertheless a good habit.