robotology-playground/WBI-Toolbox #112: Need help to understand "Using the Toolbox"

misaki43, 2016-01-12:
Hi, 
I don't understand what you explain in this part, can you help me?

1.
>What robot are you using? In the newest version (v0.2) of the toolbox you first need to define the environmental variable YARP_ROBOT_NAME in ~/.bashrc with the actual name of your robot, e.g. export YARP_ROBOT_NAME="icubGazeboSim". If you were to use the real robot, say iCubGenova03, then you assign this name to YARP_ROBOT_NAME. Rationale: Since now yarpWholeBodyInterface uses the ResourceFinder to find configuration files in your system, one of the data directories where it will search is robots/$YARP_ROBOT_NAME. Therefore, after you install yarpWholeBodyInterface, the default configuration files in it will be copied to CODYCO_SUPERBUILD_DIR/install/share/codyco/robots where you will find all the available robots. :warning: Please note that if you compile yarpWholeBodyInterface (by say, doing make in $CODYCO_SUPERBUILD_DIR/libraries/yarpWholeBodyInterface) with the flag CODYCO_INSTALL_ALL_ROBOTS=OFF and no environmental variable YARP_ROBOT_NAME defined, you won't have installed any of these configuration files. For more info: http://eris.liralab.it/yarpdoc/resource_finder_spec.html. This allows the WBI-Toolbox to be used not only with iCubGenova0X. It has been tested so far with iCubHeidelberg01 and COMAN.

I don't really understand: Need I to install "yarpWholeBodyInterface" or did you just inform us that we need to change de flag IF we need to install it? (In doubt, I installed it).

>Creating a model Before using or creating a new model keep in mind that WBI-Toolbox is discrete in principle and your simulation should be discrete as well. By going to Simulation > Configuration Parameters > Solver you should change the solver options to Fixed Step and use a discrete (no continuous states) solver.

I don't really understand this part too, but I tried to follow these instructions. The problem is I don't find this "Simulation" tab. I think I don't follow all steps correctly, could you tell me where I am wrong?
a. I added export YARP_ROBOT_NAME="icubGazeboSim" in bashrc
b. I launched yarpserver, then gazebo. I did: insert>Icub.
Then, i don't find where I can create a model (where is simulation )?
To be honest I even don't know what it is but I thought I will see it directly by test.

Thanks!

traversaro, 2016-01-12:
> I don't really understand: Need I to install "yarpWholeBodyInterface" or did you just inform us that we need to change de flag IF we need to install it? (In doubt, I installed it).

sdadas

> I don't really understand this part too, but I tried to follow these instructions. The problem is I don't find this "Simulation" tab. I think I don't follow all steps correctly, could you tell me where I am wrong?
a. I added export YARP_ROBOT_NAME="icubGazeboSim" in bashrc
b. I launched yarpserver, then gazebo. I did: insert>Icub.
Then, i don't find where I can create a model (where is simulation )?
To be honest I even don't know what it is but I thought I will see it directly by test.

Just to clarify: `Simulation > Configuration Parameters > Solver` is the path of an option in the Simulink menu, not in the Gazebo one.

misaki43, 2016-01-13:
Ok thanks,  but I have still a lot of problem, I think I miss a tutorial about this toolbox because I don't understand how it works. For example:

1) I tried to launch the torquebalancing module (following https://github.com/robotology/codyco-modules/blob/master/src/modules/torqueBalancing/README.md) but it didn't works. I did:

> yarpserver
gazebo (where I added icub feet fixed)
yarpmotorgui --from homePoseBalancingTwoFeet.ini --name icubGazeboSim 
Cliked on the "home all" button 

But then an error appears;
> No calibrator device was configured to perform this action, pleade verify that the wrapper config file for part head has the "Calibrator" keyword in the attach phrase

2) I don't understand how I can create simulation (but I don't need it now). 
3) I don't know how to link matlab files with yarp.
In fact I don't understand all things about this toolbox. Did I miss a tutorial somewhere? Could you guide me please?

traversaro, 2016-01-13:
> Ok thanks, but I have still a lot of problem, I think I miss a tutorial about this toolbox because I don't understand how it works.

It is also perfectly possible that the documentation is not appropriately describing some aspects of the software. : ) 
Please consider that this is far from being professional-level software, as it is **research** software written and **documented** by PhD students. We are happy to improve the quality of the software as much as possible, but expecting all the software to be fully documented as it is written may be a bit unrealistic. 

Regarding the errors: 
* Is the iCub in Gazebo loading correctly (i.e. it is not collapsing?). There was a bug in `gazebo-yarp-plugins` that was recently introduced (in https://github.com/robotology/gazebo-yarp-plugins/commit/844333aec783fb5189af3ed90ddd198d0968711e) and I just fixed it. So perhaps if that was the case you may want to update the `gazebo-yarp-plugins`.

> 2) I don't understand how I can create simulation (but I don't need it now). 

You mean a simulation in Gazebo or in Simulink using the WBI-Toolbox?

> 3) I don't know how to link matlab files with yarp.

What you mean by link?

> In fact I don't understand all things about this toolbox. Did I miss a tutorial somewhere? Could you guide me please?

I think it may be easier for you to understand the concepts used in the WBI-Toolbox if you a concrete goal (for example a simple controller) to implement.  If you already have something similar, if you want we can discuss 
on Skype what you are missing so at IIT we try to provide what you miss. Let me know!

misaki43, 2016-01-13:
> Please consider that this is far from being professional-level software, as it is research software written and documented by PhD students. We are happy to improve the quality of the software as [...]

Yes I know that, but I'm really lost so I think I have missed something. I just finished to install all stuffs (codyco-superbuild/iCub-main/yarp/WBI_toolbox) and now I'm lost. I read some YARP and launched some tests using yarpmanager but now, I don't know what are next steps to understand all about iCub softwares. My goal is to be able to launch a controller like "torqueBalancing". 

> Is the iCub in Gazebo loading correctly (i.e. it is not collapsing?). There was a bug in gazebo-yarp-plugins that was recently introduced (in robotology/gazebo-yarp-plugins@844333a) and I just fixed it. So perhaps if that was the case you may want to update the gazebo-yarp-plugins.

I don't think there is collapsing but there is an error "ApplyWrench pluging: link neamed iCub::root_link not found" when I put the robot in gazebo.
I update this folder (with the make) but it doesn't work better. Is my "1)" part correct or did I miss something?

>You mean a simulation in Gazebo or in Simulink using the WBI-Toolbox?

In fact, I don't know what is Simulink and how it is used, but I suppose it is not important for me. 

>>3) I don't know how to link Matlab files with yarp.

>What you mean by link?

If I understand correctly, modules are Matlab files that have to be linked to the simulator to control it. Then, how to use these Matlab files? 

>Yes I know that, but I'm really lost so I think I have missed something. I just finish to install all stuffs (codyco-superbuild/iCub-main/yarp/WBI_toolbox) and now I'm lost. I read some YARP and launched some tests using yarpmanager but now I don't know what are next steps to understand all about iCub softwares. My goal is to be able to launch a controller like "torqueBalancing". 

>>I think it may be easier for you to understand the concepts used in the WBI-Toolbox if you a concrete goal (for example a simple controller) to implement. If you already have something similar, if you want we can discuss
on Skype what you are missing so at IIT we try to provide what you miss. Let me know!

My first step is only to run the "torqueBalancing" controller. In regards to Skype, it would be a pleasure, there is my skype name : misaki.soma (I have a meeting right now but after I can)

misaki43, 2016-01-14:
Hi! 
Did you have some trouble to find me on Skype? I have done a little progress. When I launched "yarpserver"/"gazebo" (+added iCub fixed) and then:
>  yarpmotorgui --from homePoseBalancingTwoFeet.ini --name icubGazeboSim

It didn't find the “.ini”, so I had to modify it with: 
>  yarpmotorgui --from ~/codyco-superbuild/build/install/share/codyco/robots/icubGazeboSim/homePoseBalancingTwoFeet.ini --name icubGazeboSim 

Since, in the tutorial, we only have to do “ --from nameFile.ini”, I suppose I have a problem with path definition?
It may be linked with this error: in codyco-superbuild, when I do the “ccmake” I can see that the “WBIToolbox_DIR” isn't correct because the associated path is “WBIToolbox_DIR-NOTFOUND”  (It is the same for WBIToolbowControllers_DIR and WBToolbox_DIR), how can I fix it?

Now, when I click on the "home all" button it is working, so I tried to go ahead. I launched Matlab and  ran the "initTorqueBalancing.m". Then, the following warning occurred (and I don't understand this path management):
>Warning: Function
/home/odermy/Software/src/drake-v0.9.11-linux/drake/util/dev/atan2.m
has the same name as a MATLAB builtin. We suggest you
rename the function to avoid a potential name
conflict. 
In path at 110
In addpath at 87
In initTorqueBalancing at 22 

Ignoring this warning, I launched "torqueBalancing.mdl", but then Matlab crashed after many warnings. I write here, only the firsts of them:
>Warning: Loading model 'torqueBalancing' generated
with a newer  version (8.6) of Simulink 
  In general/private/openmdl at 13
  In open at 159
  In uiopen at 196 
Warning: torqueBalancing.mdl, line 101: block_diagram
does not have a parameter named
'ShowVisualizeInsertedRTB' 
  In general/private/openmdl at 13
  In open at 159
  In uiopen at 196 
Warning: torqueBalancing.mdl, line 102: block_diagram
does not have a parameter named 'ShowMarkup' 
  In general/private/openmdl at 13
  In open at 159
  In uiopen at 196 
Warning: torqueBalancing.mdl, line 204: Parameter
"SolverInfoToggleStatus" was not expected here 
  In general/private/openmdl at 13
  In open at 159
  In uiopen at 196 

Did the first warning explain that I haven't the great Simulink Version? I don't get it because it is installed with codyco-superbuild isn't it? (And I do the "pull" and the "make" even today).

traversaro, 2016-01-14:
> Did you have some trouble to find me on Skype? 

Sorry, today was a rather busy day. : ) 

> It didn't find the “.ini”
> Since, in the tutorial, we only have to do “ --from nameFile.ini”, I suppose I have a problem with path definition?

Are you sure that the `YARP_DATA_DIRS` environmental variable is including also the directories related to the codyco software (as in https://github.com/robotology/codyco-superbuild#configure-your-environment-2)?

> It may be linked with this error: in codyco-superbuild, when I do the “ccmake” I can see that the “WBIToolbox_DIR” isn't correct because the associated path is “WBIToolbox_DIR-NOTFOUND”  (It is the same for WBIToolbowControllers_DIR and WBToolbox_DIR), how can I fix it?

No, this is normal.. it means that the superbuild did not find the project in the computer and will download and build it itself. It definitely needs to be explained in the README because you are not the first user to be confused by this. 

> Did the first warning explain that I haven't the great Simulink Version?

To launch the simulink controller, please follow the steps listed here: https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers/torqueBalancing#launch-procedure .
In general, I strongly suggest you to understand what Simulink is with some simple examples before trying to understand a complex model like a balancing controller for a 23-DOFs humanoid. : ) 
The MathWorks online documentation on Simulink is quite good, check it out.

traversaro, 2016-01-14:
@misaki43 sorry I have accidentally deleted you comment... : ( anyway my reply was saved.

traversaro, 2016-01-14:
@misaki43 comment restored : )

misaki43, 2016-01-15:
>Are you sure that the YARP_DATA_DIRS environmental variable is including also the directories related to the codyco software (as in https://github.com/robotology/codyco-superbuild#configure-your-environment-2)?

Yes, that was the problem: because of a previous problem, I have commented the line "export YARP_DATA_DIRS=$YARP_DATA_DIRS:$CODYCO_SUPERBUILD_ROOT/build/install/share/codyco" and forgot to uncomment it after.

>To launch the simulink controller, please follow the steps listed here: https://github.com/robotology-playground/WBI-Toolbox-controllers/tree/master/controllers/torqueBalancing#launch-procedure .
In general, I strongly suggest you to understand what Simulink is with some simple examples before trying to understand a complex model like a balancing controller for a 23-DOFs humanoid. : )
The MathWorks online documentation on Simulink is quite good, check it out. 

You are right, I'm going to do Simulink's tutorials, but in the same time I want to resolve all problems to directly move on after. 
And when I follow steps written in your link, I have the same trouble (warnings following by Matlab crash).
It is the last problem I think, can you help me? (Thanks, you have already help me a lot).

misaki43, 2016-01-20:
My problem is maybe linked to https://github.com/robotology-playground/WBI-Toolbox-controllers/issues/13 because before warnings I can read 
>Warning: Loading model 'torqueBalancing'
generated with a newer  version (8.6) of
Simulink

traversaro, 2016-01-20:
Ok, then we will wait on @DanielePucci  feedback.

DanielePucci, 2016-01-21:
Exported to MATLAB201b. The tests I performed were ok. Let us know.

misaki43, 2016-01-26:
Hi ! So now, it is working ! I have read some Simulink tutorial (even though I am still a beginner) and I have read the "simulink code". Thanks to them, I approximately understand how the controller works (even though functions like "controller FCN" are still too hard for me to understand).

I just have a few final questions:

1. In the function "Read desired_x_dx_ddx_CoM", we can directly give some desired goals, is that correct? If that is the case, how can I connect it and communicate with it? (Same for Read qDes_impedences).
I tried to check the "autoconnect" option for these two functions and then to modify different positions with "yarpmotorgui": is that correct too?

2. Are "generalized bias forces" of the same kind as the Coriolis one? (Why could I not read the function robotState?)

3. In the CoDyCo video, we can see that the robot is able to raise one foot and at the same time to maintain its equilibrium. So how come the robot falls when I apply forces on one foot to raise it? (I modified the constant named "LEFT_RIGHT_FOOT_IN_CONTACT" to allow its left foot to be raised, but it didn't work).